naugehyde/bt-sensors-plugin-sk #1: v1.0.2 - failed to start

Haimana, 2024-09-06:
Hello,

Thank you for your work.
Unfortunately, the plugin if fails to start after updating to v1.0.2. Running on Pi5/OP4.

Best regards,
Seb

> 
> Sep 06 07:05:38 bt-sensors-plugin-sk failed to start: Cannot find module './node_modules/node-ble/src/Device.js' Require stack: - /home/pi/.signalk/node_modules/bt-sensors-plugin-sk/index.js - /usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js - /usr/lib/node_modules/signalk-server/lib/interfaces/index.js - /usr/lib/node_modules/signalk-server/lib/serverroutes.js - /usr/lib/node_modules/signalk-server/lib/index.js - /usr/lib/node_modules/signalk-server/bin/signalk-server
> Sep 06 07:05:38 Error: Cannot find module './node_modules/node-ble/src/Device.js' Require stack: - /home/pi/.signalk/node_modules/bt-sensors-plugin-sk/index.js - /usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js - /usr/lib/node_modules/signalk-server/lib/interfaces/index.js - /usr/lib/node_modules/signalk-server/lib/serverroutes.js - /usr/lib/node_modules/signalk-server/lib/index.js - /usr/lib/node_modules/signalk-server/bin/signalk-server at Module._resolveFilename (node:internal/modules/cjs/loader:1077:15) at Module._load (node:internal/modules/cjs/loader:922:27) at Module.require (node:internal/modules/cjs/loader:1143:19) at require (node:internal/modules/cjs/helpers:121:18) at Object.<anonymous> (/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/index.js:9:16) at Module._compile (node:internal/modules/cjs/loader:1256:14) at Module._extensions..js (node:internal/modules/cjs/loader:1310:10) at Module.load (node:internal/modules/cjs/loader:1119:32) at Module._load (node:internal/modules/cjs/loader:960:12) at Module.require (node:internal/modules/cjs/loader:1143:19) at require (node:internal/modules/cjs/helpers:121:18) at doRegisterPlugin (/usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js:397:39) at registerPlugin (/usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js:258:13) at /usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js:209:13 at Array.forEach (<anonymous>) at startPlugins (/usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js:208:85) at Object.start (/usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js:46:13) at /usr/lib/node_modules/signalk-server/lib/index.js:503:50 at /usr/lib/node_modules/signalk-server/node_modules/lodash/lodash.js:4967:15 at Function.forIn (/usr/lib/node_modules/signalk-server/node_modules/lodash/lodash.js:13018:11) at startInterfaces (/usr/lib/node_modules/signalk-server/lib/index.js:480:22) at Server.<anonymous> (/usr/lib/node_modules/signalk-server/lib/index.js:328:17) { code: 'MODULE_NOT_FOUND', requireStack: [ '/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/index.js', '/usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js', '/usr/lib/node_modules/signalk-server/lib/interfaces/index.js', '/usr/lib/node_modules/signalk-server/lib/serverroutes.js', '/usr/lib/node_modules/signalk-server/lib/index.js', '/usr/lib/node_modules/signalk-server/bin/signalk-server' ] }
> 
> 
>

naugehyde, 2024-09-06:
Thanks for letting me know.

Can you send me a directory listing of your '/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/'?

The stack trace is saying the node-ble package isn't installed there. And I don't understand why it isn't.

Haimana, 2024-09-06:
[image]

naugehyde, 2024-09-06:
The problem is the required node-ble package was never installed.

How did you install the plugin? From the app store?

Haimana, 2024-09-06:
yes, from the app store.
it worked before this update

naugehyde, 2024-09-06:
very strange.

not sure what happened there. there's no node_modules directory under the plugin.

can you run 'npm i' from '/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/' and send any errors or the directory listing after running?

thanks.

Haimana, 2024-09-06:
pi@haimana:~/.signalk/node_modules/bt-sensors-plugin-sk $ npm i
npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm warn deprecated har-validator@5.1.5: this library is no longer supported
npm warn deprecated npmlog@4.1.2: This package is no longer supported.
npm warn deprecated rimraf@3.0.2: Rimraf versions prior to v4 are no longer supported
npm warn deprecated are-we-there-yet@1.1.7: This package is no longer supported.
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm warn deprecated gauge@2.7.4: This package is no longer supported.
npm warn deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142

added 29 packages, and audited 30 packages in 3s

5 packages are looking for funding
  run `npm fund` for details

7 moderate severity vulnerabilities

To address all issues (including breaking changes), run:
  npm audit fix --force

Run `npm audit` for details.
pi@haimana:~/.signalk/node_modules/bt-sensors-plugin-sk $

naugehyde, 2024-09-06:
Well, despite all the warnings, node-ble (the support package for talking to Bluetooth) did install. Your plugin should work now. 

Just don't know why node-ble didn't install in the first place.

Haimana, 2024-09-06:
and after that "node_modules" appear inside folder
it seems the plugin is working now, at least it start without error

pi@haimana:~/.signalk/node_modules/bt-sensors-plugin-sk $ ls -l
total 96
-rw-r--r--  1 pi pi  1952 Sep  6 08:11 BTSensor.js
-rw-r--r--  1 pi pi  7497 Sep  6 08:11 index.js
drwxr-xr-x 31 pi pi  4096 Sep  6 08:13 node_modules
-rw-r--r--  1 pi pi   696 Sep  6 08:11 package.json
-rw-r--r--  1 pi pi 53517 Sep  6 08:13 package-lock.json
-rw-r--r--  1 pi pi 12961 Sep  6 08:11 README.md
drwxr-xr-x  2 pi pi  4096 Sep  6 08:11 sensor_classes

Haimana, 2024-09-06:
yes, it seems to work.
thank you for your support.

now, I'll try to adapt it for reading a junctek kl140f shunt, but this is another story

thank you again

naugehyde, 2024-09-06:
Good luck! Let me know how it goes. Never heard of that product before but if you can get it to work I can put your class into the repository.

Haimana, 2024-09-06:
Thank you,

Maybe next week i'll find more time for this. I'll keep you updated.

Fair winds

closed 2024-09-06 by Haimana

naugehyde, 2024-09-06:
Check out:

https://github.com/AnalogThinker/junctek_monitor

Fair winds!