naugehyde/bt-sensors-plugin-sk #2: Polling frequency

sveinhansen, 2024-09-13:
Hi

I was pleased to see this plugin for SignalK.

Would it be an idea to be able to configure a polling frequency for a BLE device?

I suspect that polling it often will drain the battery of the Xiaomi LYWSD03MMC device. I use these devices to monitor temp/humid during the winter i Norway to ensure that I do not have too much condensation etc. I typically need 1-2 hours frequency for sampling the data.

Thanks!

Regards,
Svein

naugehyde, 2024-09-20:
Svein,

As of today, I have a working pre-beta that includes a polling option for the LYWSD03MMC. If you want to test it, let me know and I'll walk you through setting it up.

-Andrew

SCarns, 2024-09-30:
What are the other devices you're planning to support? I'd like to vote for RuuviTags first, as that is the easiest, I believe, to implement. I just not enough of a programmer to create a new class, although I'm researching it. I also am looking to add a few Victron devices (BMV712, SmartCharger IP43, and a SmartSolar Charger MPPT.

I get how the data from the RuuviTag is transmitted, I'm just uncertain how to parse that data and make a new class with it. I really wish I could be of more help here...

naugehyde, 2024-09-30:
RuuviTags are on my list.

Will let you know when it's available -- I'll need testers as I don't have the device myself!

naugehyde, 2024-09-30:
Also, plan on supporting as much of Victron's product line that can be supported. See the 1.1.0 tree for a list (under sensor_ classes).

SCarns, 2024-09-30:
I have 6 RuuviTags. I'm happy to test for you!

naugehyde, 2024-09-30:
No prob. Will let you know when the Beta (Alpha?) version with RuuviTags is ready.

naugehyde, 2024-10-06:
Hi Svein. Support for polling frequency (for all GATT capable sensors like the Xiaomi LYWSD03MMC) and preliminary, bench-tested RuuviTag, BMV712, Solar Charger and SmartCharger support is in the repository. 

To get the beta installed, follow the instructions in the readme for installing from github. After that, there is one extra step for getting the beta:

```
cd ~/.signalk/node_modules/bt_sensors_plugin_sk
git switch "1.1.0"
```
Let me know if you need help in installing and if you have any problems or questions with the plugin. 

To update the code until the new version is in the repository (I'm still tinkering), just do a "git pull" from the install directory.

It's very helpful to have you testing this, so thank you.

SCarns, 2024-10-07:
A follow up on this:

I installed and the plugin found 17 devices. I recognized my BMV-712 and my SmartSolar 100/50, but not my Victron IP43 smart charger nor any of my RuuviTags, although I did recognize their MAC addresses in the list. I need more time to play with it, as I didn't get any data into SignalK, but that could've been because I just threw in some deltas and didn't go all the way with victron pass keys, etc.

naugehyde, 2024-10-07:
Awesome. Not a terrible start!

Not sure why your RuuviTags are not getting properly discovered. The IP43
is also a mystery (I've only tested with a IP65/67)
.
Can you send me the output of your bluetoothctl as its scanning? Running it
for like five minutes should be fine.

Just run the following code from a command prompt:

bluetoothctl scan on >> bt_cap.log

hit ctrl-c to stop the capture after a few minutes then send the file. it'd
be a huge help.

naugehyde, 2024-10-07:
Also, I'm making frequent changes to the 1.1.0 branch so be sure to run "git pull" from the plugin's source directory so you'll always have the latest changes.

naugehyde, 2024-10-09:
Found the problem with the RuuviTags. Latest code should identify the RuuviTag properly. 

Pull down the latest revisions (git pull) and let me know if it flies.

The bluetooth capture would still be helpful in getting support for the IP43. So please send along if you have the time.

Also, I suggest you make sure the IP43 device has the latest Victron firmware -- VE only began to support encrypted BT broadcast in the last three years or so.

sveinhansen, 2024-10-10:
> @sveinhansen
> 
> Hi Svein. Support for polling frequency (for all GATT capable sensors like the Xiaomi LYWSD03MMC) and preliminary, bench-tested RuuviTag, BMV712, Solar Charger and SmartCharger support is in the repository.
> 
> To get the beta installed, follow the instructions in the readme for installing from github. After that, there is one extra step for getting the beta:
> 
> ```
> cd ~/.signalk/node_modules/bt_sensors_plugin_sk
> git switch "1.1.0"
> ```
> 
> Let me know if you need help in installing and if you have any problems or questions with the plugin.
> 
> To update the code until the new version is in the repository (I'm still tinkering), just do a "git pull" from the install directory.
> 
> It's very helpful to have you testing this, so thank you.

Hi there.

Thanks for taking my request for polling frequency. I'm afraid I'm not able to test it during this fall/winter, but will for sure start using the plugin from spring time. 

I will convert my own setup to this plugin later.

Currently, I'm collecting data from a Vicrton MPPT and BMV-712 though ve.direct over usb to signalk ve-direct plugin.. 

For the Xiaomi I wrote a Python script to poll from the sensors triggered by a @hourly cron-job, pushing the values to signalk on the udp-interface of signalk.

With your plugin, I can get rid of the ve.direct -> usb cables, my cron jobs and scripts to call and decifer the xiaomi sensors. 

Again, thanks!

-Svein

SCarns, 2024-10-13:
Finally made it back to the boat to work with this. RuuviTags work great! All six showed up in the drop-down menu and each was easily assigned and is very stable. I like that I don't have to fill out all the SK delta fields, as I don't need x|y|z, etc. data in my set up. 

The Victron stuff, however, I'm struggling with. I installed the BMV-712 and assigned the appropriate deltas and it immediately started pumping data into SK. When I went to add in the MPPT 100/50, it caused the BMV stop working. I assume the answer to this will show up in the bluetoothctl log, but I ran out of time and couldn't pull it from the RPi. A few more days and I'll have it to you.

This is a really great piece work. I'm sure this will all get worked out, but so far, I love it! I really simplifies my set up while doing everything I need. Thanks for your ideas and efforts!!!

SCarns, 2024-10-13:
Upate: I just checked SK Cloud and my BMV data is showing up. That's interesting. I'll add in the MPPT Solar next time I get there and see what we can get out of it.

naugehyde, 2024-10-13:
Cool! Happy to hear the plugin is working (more or less).

FYI when you add a new device and hit submit, the plugin restarts from
scratch, and reconnects to all active devices. Which is probably why the
BMV appeared offline temporarily.

Is the IP43 working? I don't have my hands on one so I couldn't test before
releasing. I was able to test the IP65 (minimally). Just don't know if the
IP43 and IP65 are the same class of sensor. Can't find any docs from
Victron on it.

Just to be safe, can you enable debug on the server and send me the log?

Out of curiosity, does SK Cloud store the log and allow sharing? That would
make remote debugging a little easier.

News! The plugin beta is in the NPM repository-- not announced yet and docs
are not up to date.

Best,

Andrew Gerngross

SCarns, 2024-10-13:
I don't think SK Cloud is that high tech. My SK, and many others, dump data to it. It's real value is I can use in the WilhelmSK and view my gauges remotely. No logging is available.

I hope to make it back to the boat tmro. I'll try the IP43 then. Stay tuned!

SCarns, 2024-10-15:
IP43 does not work. The plugin sees it, but does not identify it at a charging device - I just get the "name" and a single SK path field.

I had the BMV-712 and the Smart Solar MPPT 100/50 working great today with the IP43 off. When I turned it back on, it all failed. IP43 doesn't have and encryption key. I ran out of time again. Next time, I will remove all else and try just installing the IP43 to see if it will go.

naugehyde, 2024-10-15:
Did you send the bluetooth capture? If not, please do. It's the best way
for me to debug from a distance. Also, turn the plugin debug on and send me
the relevant pieces of your log when the IP43 is activated. It shouldn't
bring down the other devices. I'll look and fix ASAP.

SCarns, 2024-10-15:
[bt_cap.log](https://github.com/user-attachments/files/17381945/bt_cap.log)

SCarns, 2024-10-15:
[skserver-raw_2024-10-14T19.log](https://github.com/user-attachments/files/17382057/skserver-raw_2024-10-14T19.log)
[skserver-raw_2024-10-14T20.log](https://github.com/user-attachments/files/17382058/skserver-raw_2024-10-14T20.log)
[skserver-raw_2024-10-14T21.log](https://github.com/user-attachments/files/17382059/skserver-raw_2024-10-14T21.log)
[skserver-raw_2024-10-14T22.log](https://github.com/user-attachments/files/17382061/skserver-raw_2024-10-14T22.log)
[skserver-raw_2024-10-14T23.log](https://github.com/user-attachments/files/17382062/skserver-raw_2024-10-14T23.log)
[skserver-raw_2024-10-15T00.log](https://github.com/user-attachments/files/17382064/skserver-raw_2024-10-15T00.log)
[skserver-raw_2024-10-15T01.log](https://github.com/user-attachments/files/17382066/skserver-raw_2024-10-15T01.log)
[skserver-raw_2024-10-15T02.log](https://github.com/user-attachments/files/17382067/skserver-raw_2024-10-15T02.log)
[skserver-raw_2024-10-15T03.log](https://github.com/user-attachments/files/17382068/skserver-raw_2024-10-15T03.log)
[skserver-raw_2024-10-15T04.log](https://github.com/user-attachments/files/17382069/skserver-raw_2024-10-15T04.log)
[skserver-raw_2024-10-15T05.log](https://github.com/user-attachments/files/17382070/skserver-raw_2024-10-15T05.log)
[skserver-raw_2024-10-15T06.log](https://github.com/user-attachments/files/17382071/skserver-raw_2024-10-15T06.log)
[skserver-raw_2024-10-15T07.log](https://github.com/user-attachments/files/17382072/skserver-raw_2024-10-15T07.log)
[skserver-raw_2024-10-15T08.log](https://github.com/user-attachments/files/17382074/skserver-raw_2024-10-15T08.log)
[skserver-raw_2024-10-15T09.log](https://github.com/user-attachments/files/17382075/skserver-raw_2024-10-15T09.log)
[skserver-raw_2024-10-15T10.log](https://github.com/user-attachments/files/17382077/skserver-raw_2024-10-15T10.log)
[skserver-raw_2024-10-15T11.log](https://github.com/user-attachments/files/17382078/skserver-raw_2024-10-15T11.log)
[skserver-raw_2024-10-15T12.log](https://github.com/user-attachments/files/17382079/skserver-raw_2024-10-15T12.log)
[skserver-raw_2024-10-15T13.log](https://github.com/user-attachments/files/17382081/skserver-raw_2024-10-15T13.log)
[skserver-raw_2024-10-15T14.log](https://github.com/user-attachments/files/17382082/skserver-raw_2024-10-15T14.log)
[skserver-raw_2024-10-15T15.log](https://github.com/user-attachments/files/17382085/skserver-raw_2024-10-15T15.log)
[skserver-raw_2024-10-15T16.log](https://github.com/user-attachments/files/17382086/skserver-raw_2024-10-15T16.log)
[skserver-raw_2024-10-15T17.log](https://github.com/user-attachments/files/17382087/skserver-raw_2024-10-15T17.log)

SCarns, 2024-10-15:
Did a "git pull" this morning while digging out logs for you and here's the data SK is getting from the plugin :

[image]

naugehyde, 2024-10-15:
Looks like the fix broke the RuuviTags. I'll release a better version ASAP.

SO hard to this without devices!

SCarns, 2024-10-15:
Where are you located. I have an extra RuuviTag I can send you.

naugehyde, 2024-10-15:
I'm guessing your IP43's mac address is D8:D6:CB:46:5F:EE.

Can you confirm exactly what happens when you activate it?

Also, I'll need the debug log, not the raw server logs. Thank you!

naugehyde, 2024-10-15:
Thank you but I got my first RuuviTag this past weekend. It's the Victron
devices that are vexing me.

SCarns, 2024-10-15:
> I'm guessing your IP43's mac address is D8:D6:CB:46:5F:EE. Can you confirm exactly what happens when you activate it? Also, I'll need the debug log, not the raw server logs. Thank you!
> […](#)

Yes, that is the one. When it is on, all the Victron devices show up in the pull-down menu, but when selected, I only get Device Parameters/Name and SignalK Path/Signal Strength options for each Victron Device. 

With it off, I've been successful and setting up BMV-712 (DF:8A:64:9B:78:06) and Solar (E0:1F:C7:3B:97:0C). When the IP43 comes back in, they all revert to the above.

naugehyde, 2024-10-15:
Weird!

Screen shots would help. Hope I'm not being a pain in the ass!

Also the IP43 does not appear to have the latest BT support. Not sure if
Victron is releasing it or not. I hear they have released betas for the IP
22 and 67 tho.

naugehyde, 2024-10-15:
Screen shots of the config page when the IP43 is active, that is.

SCarns, 2024-10-15:
I’ve left the boat (damn job), but we’ll get you whatever you need in the next couple of days when I get back to it.

I was wondering about the Bluetooth update. VictronConnect says it’s running the latest version. I’ll look into that some more tonight and reach out to Victron to ask for an update.

naugehyde, 2024-10-18:
FYI Victron Connect beta 3.61 has firmware for recent models of the IP43 that appear to be compatible with my plugin.

https://www.victronenergy.com/live/victronconnect:beta

SCarns, 2024-10-18:
I uploaded the beta onto the IP43. It now has an encryption key. Great!

RuuviTags are solid. However, the plugin doesn't do anything with any of my Victron stuff now. 
[image]

SCarns, 2024-10-18:
This is what I usually get...
[image]

SCarns, 2024-10-18:
Update: After a reboot, I had it connected to the IP43. Filled in the fields and hit submit. Getting close. I'll keep working with it.

SCarns, 2024-10-18:
Update: IP43 data is being captured and sent into SignalK. Question is can I add the other two into the mix without killing the IP43?
[image]

naugehyde, 2024-10-18:
Yes. Should work.

SCarns, 2024-10-18:
The values aren't correct and don't match up with what I see in victronconnect. Still messing with it.

SCarns, 2024-10-18:
Something I am noticing is that I can property install the device is it has it's full/correct name in the dropdown menu.
[image]
[image]

naugehyde, 2024-10-18:
Thank you.

Can you clarify which device is reporting bad data? I'll need a screenshot
or log of the data. Also, have you double checked that the encryption key
is correct? Finally, can you DM your bt-sensors-plugin-sk.json
configuration file?

SCarns, 2024-10-18:
Had to leave for work again, so I’ll try to get you that tomorrow.

Here’s my best non-technical estimation of what I think is happening here. I think there’s an initial scan where the plug-in sees “Victron BMV – 712 Smart. Rev 2 (MAC address)”, what’s the plug-in recognizes as a Victron BMV and let me set it up with the appropriate information. I can even come back and add in the starting battery values, which show up after I hit submit the first time.

According to VictronConnect, the device’s name is “Smart BMV-712 (serial number)”, which eventually gets passed on to the plug-in. At this point, the plug-in gets confused as it doesn’t recognize what it is any longer and displays is as an unknown device.

My plan tomorrow is to use VictronConnect to name the device to match it initial scanned and see if that will no longer confuse the plug-in.

naugehyde, 2024-10-18:
Name of plugin doesn't matter. Config is keyed to the MAC Address.

The config is quirky. But if you start up the plugin and leave it alone for
a minute or two, the config screen is properly set up.

Also, which version of the plugin are you using?

SCarns, 2024-10-18:
1.1.0 beta 2.1.2

naugehyde, 2024-10-18:
OK.

Definitely DM me your bt-sensors-plugin-sk.json file.

To clarify is it just the BMV-712s causing you issues right now? If the Ip43 is reporting unusual data can you send me a screen capture or log of that data?

SCarns, 2024-10-19:
Ok, I'll get that tmro.

Side question: I was looking at your code. I see that the IP43 1+1 is a different model code. Does it have a different set of fields? I got the one for the 3-bank IP43.

closed 2024-10-26 by naugehyde