naugehyde/bt-sensors-plugin-sk #3: Request for new device

Haimana, 2024-10-23:
Hello, 
Me again! :)
Congrats for your work, good job!

I am wondering how much time consuming will be for adding a new device. For now, I'm referring at Junctek Baterry Monitor, KG140F.
http://68.168.132.244/KG-F_EN_manual.pdf
And some related repos:
https://github.com/AnalogThinker/junctek_monitor
https://github.com/mysystem32/juntek_kg

I have one of this on my boat and I can do tests. Unfortunately, for now, I have only remote access, through Pi. So will be difficult to collect some logs with the cellphone.
Your plugin detect it and report it's signal strength, so some connection is made.
[image]
[image]

Also, I can try to add it myself, even I am not a real programmer. Just give me some hints/guidelines about how to do that.
Thank you

Best regards,
Seb

naugehyde, 2024-10-24:
Hi Seb!

I'll take a look at it when I'm done with Victron devices.

It would be tough for a non-programmer to implement without clear manufacturer specs. You'd have to sift through the python code linked above for clues as to what the protocol and data structures are. Not impossible but it can be confounding.

I'll get to it down the road. In the meantime you can download and run nrf conenct https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_US an app which interrogates, connects and displays the services of a bluetooth device. I'll leave it up to you to figure out how to use it. If you do get around to it and contact your device, send me screenshots of the device's service characteristics. It'll help.

Best,

Andrew Gerngross

Haimana, 2024-10-24:
Thanks, take your time 
I'm also in the middle of some work and far away from boat. 
I'll update when I'll be there and scan for characteristics and other details
Best regards

Haimana, 2024-12-17:
hello again,
I tried to make a BT capture with my phone.
Unfortunately I'm not sure if what I got is OK for finding useful info regarding this device
I'll attach the extracted file (with adb), found in FS\data\misc\bluetooth\logs
[BT_HCI_2024_1215_164527.cfa.zip](https://github.com/user-attachments/files/18162138/BT_HCI_2024_1215_164527.cfa.zip)

Can you decode this?
Thank you, 
Seb

naugehyde, 2024-12-17:
This is a good start.

It would also be useful to get the bluetoothctl logs.

From a command line run the following:

```
bluetoothctl info 38:3B:26:5A:6B:F4 > seb_bt.cap
bluetoothctl scan on >> seb_bt.cap

```

the last one will run until you stop it so hit ctrl-c after about 2 minutes then dm me the seb_bt.cap file.

Haimana, 2024-12-17:
Hello,
and thank you for your support
this is my .cap file:
[seb_bt.zip](https://github.com/user-attachments/files/18170624/seb_bt.zip)
Thanks

naugehyde, 2025-03-17:
Apologies for the delay in getting to this. 

Can you install and confirm if the following works: https://github.com/AnalogThinker/junctek_monitor

If so, I can use it as a starting point for developing a device class for the plugin.

Thanks!

Haimana, 2025-03-19:
Hello, thanks for support. 
And don't worry about delay. I'm not rushing with this. 
I'll do the tests and report back ASAP.
Thank you

Haimana, 2025-04-14:
hello again
I did some tests under xubuntu, laptop install
I removed the mqtt related lines from script
this is what i get:

```
venv) seb@ideapad:~/_GitHub_/junctek_monitor$ python3 BTjuntek.py --debug
Setting logging level to DEBUG
Starting up
Located JUNTEK device - name=BTG039 addr=38:3B:26:5A:6B:F4
DEBUG: DATA=bb23652700d575ee
{}
DEBUG: DATA=bb23652701d576ee
{}
DEBUG: DATA=bb23652702d577ee
{}
(venv) seb@ideapad:~/_GitHub_/junctek_monitor$ python3 BTjuntek.py --debug
Setting logging level to DEBUG
Starting up
Located JUNTEK device - name=BTG039 addr=38:3B:26:5A:6B:F4
DEBUG: DATA=bb23652746d545eebb1337c053ee
{'voltage': 13.37}
DEBUG: DATA=bb23652747d546ee
{}
DEBUG: DATA=bb23652748d547ee
{}
DEBUG: DATA=bb23652749d548ee
{}
(venv) seb@ideapad:~/_GitHub_/junctek_monitor$ python3 BTjuntek.py --debug
Setting logging level to DEBUG
Starting up
Located JUNTEK device - name=BTG039 addr=38:3B:26:5A:6B:F4
DEBUG: DATA=bb09c10120d838ee
{'power': 1.2, 'current': 0.09}
DEBUG: DATA=bb23652764d575ee
{}
DEBUG: DATA=bb23652765d576ee
{}
DEBUG: DATA=bb1338c054ee
{'voltage': 13.38}
DEBUG: DATA=bb23652766d577ee
{}
DEBUG: DATA=bb00c100d896ee
{'power': 0.0, 'current': 0.0}
(venv) seb@ideapad:~/_GitHub_/junctek_monitor$ 
```

it seems the data is not very consistent but I think could be a start. 
thank you,
Seb

Haimana, 2025-04-14:
and some blueman reports:

[image]

`
00001800-0000-1000-8000-00805f9b34fb Generic Access

00001801-0000-1000-8000-00805f9b34fb Generic Attribute

0000180a-0000-1000-8000-00805f9b34fb Device Information

0000ffe0-0000-1000-8000-00805f9b34fb Unknown

0000fff0-0000-1000-8000-00805f9b34fb Unknown

bluetooth:v0739p0000d0110

/org/bluez/hci0

{43863: [7, 1, 67, 72, 57, 49, 52, 49], 57100: [49, 0, 0, 0, 0, 0]}

`

Haimana, 2025-06-19:
Hello, 
Sorry for my late reply, I'm in the middle of some other project which keep me busy. 
Regarding the inconsistency, I don't mean the fluctuation of the voltage and so on. 
I mean some incomplete readings, from time to time. 
Not every reading hold the complete sentence. 
At least this is my feeling, nothing sure for now.
Anyway, thank you for not forget about me and my device. Really appreciate it. 
Thank you for your nice work, too 
Best regards, 
Seb

naugehyde, 2025-06-19:
If you want to try it sooner add the attached to your ~/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes and rename it to Junctek.js  

Not confident it'll work right away but together we can debug it.

[Junctek.js.txt](https://github.com/user-attachments/files/20811688/Junctek.js.txt)

Haimana, 2025-06-21:
Hello, 

I tried it but it seems the data is not present in SK data browser. 
This is the SK log: 

Dashboard

Webapps

Data Browser

Appstore

Server

Settings

Data Connections

Plugin Config

Server Logs

Update

Data Fiddler

Backup/Restore

Security

Documentation

OpenApi

Server Log

bt-sensors-plugin-sk

signalk-to-noforeignland

Select the appropriate debug keys to activate debug logging for various components on the server.

Persist debug settings over server restarts 

Pause the log window 

Jun 21 07:47:37 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 3.495 ms - 23
Jun 21 07:47:39 GET /signalk/v2/api/resources/notes?position=[28.547854787945262,43.80435674851512]&distance=185000 200 2.026 ms - 2
Jun 21 07:47:39 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 7.358 ms - 23
Jun 21 07:47:41 GET /@signalk/freeboard-sk/assets/img/ship_red.png 304 1.623 ms - -
Jun 21 07:47:41 GET /signalk/v2/api/resources/notes?position=[28.547854787945262,43.80435674851512]&distance=185000 304 13.738 ms - -
Jun 21 07:47:41 GET /signalk/v2/api/resources/regions?position=[28.547854787945262,43.80435674851512]&distance=185000 200 12.113 ms - 2
Jun 21 07:47:41 GET /signalk/v2/api/resources/regions?position=[28.547854787945262,43.80435674851512]&distance=185000 304 1.795 ms - -
Jun 21 07:47:42 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 3.512 ms - 23
Jun 21 07:47:43 GET /@signalk/freeboard-sk/assets/img/atons/real-south.svg 304 1.407 ms - -
Jun 21 07:47:43 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 5.439 ms - 23
Jun 21 07:47:44 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 6.000 ms - 23
Jun 21 07:47:46 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 7.397 ms - 23
Jun 21 07:48:02 GET /signalk/v2/api/resources/notes?position=[28.54691657704763,43.79943087867707]&distance=185000 200 8.201 ms - 2
Jun 21 07:48:02 Could not parse applicationData:Unexpected end of JSON input
Jun 21 07:48:02 GET /signalk/v2/api/resources/notes?position=[28.565116080967808,43.78829200763502]&distance=185000 200 1.996 ms - 2
Jun 21 07:48:02 Could not parse applicationData:Unexpected non-whitespace character after JSON at position 3210
Jun 21 07:48:02 Could not parse applicationData:Unexpected end of JSON input
Jun 21 07:48:02 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 29.856 ms - 23
Jun 21 07:48:02 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 24.806 ms - 23
Jun 21 07:48:02 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 17.646 ms - 23
Jun 21 07:48:02 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 12.674 ms - 23
Jun 21 07:48:02 GET /signalk/v2/api/resources/notes?position=[28.593687269900126,43.74993948447411]&distance=185000 200 3.664 ms - 2
Jun 21 07:48:02 POST /signalk/v1/applicationData/user/freeboard/1.0.0/ 200 8.586 ms - 23
[... 59 lines cut ...]
Jun 21 08:23:23 GET /@signalk/freeboard-sk/assets/icons/icon-72x72.png 304 3.316 ms - -
Jun 21 08:23:23 GET /@signalk/instrumentpanel/icons/instrumentpanel-72x72.png 304 2.142 ms - -
Jun 21 08:23:23 GET /@mxtommy/kip/assets/icon-72x72.png 304 1.521 ms - -
Jun 21 08:23:23 GET /@signalk/signalk-node-red/node-red-icon-2.png 304 1.484 ms - -
Jun 21 08:23:33 GET /skServer/plugins 304 8.572 ms - -
Jun 21 08:23:33 GET /bt-sensors-plugin-sk/681.js 304 3.098 ms - -
Jun 21 08:23:33 GET /bt-sensors-plugin-sk/847.js 304 1.995 ms - -
Jun 21 08:23:33 GET /plugins/bt-sensors-plugin-sk/getPluginState 200 1.767 ms - 48
Jun 21 08:23:33 GET /plugins/bt-sensors-plugin-sk/getDomains 304 6.953 ms - -
Jun 21 08:23:33 2025-06-21T07:23:33.814Z bt-sensors-plugin-sk Sending sensors
Jun 21 08:23:33 GET /plugins/bt-sensors-plugin-sk/getBaseData 304 1.868 ms - -
Jun 21 08:23:33 GET /plugins/bt-sensors-plugin-sk/getSensors 200 2.166 ms - -
Jun 21 08:23:33 2025-06-21T07:23:33.820Z bt-sensors-plugin-sk Sending progress
Jun 21 08:23:33 GET /plugins/bt-sensors-plugin-sk/getProgress 304 1.851 ms - -
Jun 21 08:23:38 GET /plugins/bt-sensors-plugin-sk/sse 200 2.861 ms - -
Jun 21 08:23:38 2025-06-21T07:23:38.827Z bt-sensors-plugin-sk deregistering session
Jun 21 08:23:41 GET /skServer/logfiles/ 304 4.726 ms - -
Jun 21 08:23:41 GET /skServer/debugKeys 200 1.253 ms - -

 

The single data present in SK browser is the BT signal level. 
The device is present in configured and electrical tabs, which looks normal for me

Thank you

Haimana, 2025-06-21:
[image]

naugehyde, 2025-06-21:
1) turn off access logging. It clogs up the server logs. To do this, go to Server->settings. There's a button to turn off accessLogging. 
2) restart SignalK

The problems with the Junctek will be easier to find in the logs after access logging is turned off.

Haimana, 2025-07-03:
Hello,
Sorry for my late reply.
Unfortunately no good news regarding Junctek... Latest plugin version installed.
I still have no data from the device (just the BT signal strength)

this is what I have in the logs:
```
Jul 03 10:29:47 2025-07-03T09:29:47.475Z bt-sensors-plugin-sk Loading plugin 1.2.3
Jul 03 10:29:47 2025-07-03T09:29:47.501Z bt-sensors-plugin-sk Connecting to bluetooth adapter hci0
Jul 03 10:29:47 signalk-server running at 0.0.0.0:[object Object]
Jul 03 10:29:47 2025-07-03T09:29:47.708Z bt-sensors-plugin-sk Starting scan...
Jul 03 10:29:47 2025-07-03T09:29:47.708Z bt-sensors-plugin-sk Setting Bluetooth transport option to auto. DuplicateData to false
Jul 03 10:29:47 2025-07-03T09:29:47.713Z bt-sensors-plugin-sk Plugin 1.2.3 started
Jul 03 10:29:47 2025-07-03T09:29:47.715Z bt-sensors-plugin-sk { Discovering: Variant { signature: 'b', value: true } }
Jul 03 10:29:48 [object Object]
Jul 03 10:29:48 2025-07-03T09:29:48.704Z bt-sensors-plugin-sk adding sensor to list 38:3B:26:5A:6B:F4
Jul 03 10:29:48 2025-07-03T09:29:48.721Z bt-sensors-plugin-sk Paths activated for BTG039 (38:3B:26:5A:6B:F4 RSSI: -58 db / 71%) ▂ ▄ ▆ 
```

I can give you access in my Pi (wireguard vpn), if will be easier for you. Contact me at sebba at end dot ro
Thank you

naugehyde, 2025-08-22:
Closing (latently). See: https://github.com/naugehyde/bt-sensors-plugin-sk/blob/main/sensor_classes/Junctek.js

closed 2025-08-22 by naugehyde