naugehyde/bt-sensors-plugin-sk #4: Victron SmartSolar testing

Peter-Petrik, 2024-10-24:
Per the references in #2 and the latest available v1.1.0-beta.2.1.3.3 through SignalK store, I managed to get 3 SmartSolar devices connected, as well as a VictronSmartBatteryProtect (Model ID:41905).

It took multitude of attempts to get the devices recognized at first. Often, I would not see the devices, or after adding I'd see only ability to add "Name of sensor" and "Signal strength in db". No ability to add Encryption Key, no additional Signalk Paths. A few restarts, disable/re-enable to plugin, and even a RPi reboot later, I got all 4 devices added and they appear to be reporting.

Glad to test further as needed and provide logs per instructions if that helps elsewhere.

One interesting (useful?) observation, upon restarting SignalK, I frequently see:

`Sensor at XX:XX:XX:XX:XX:XX unavailable. Reason: Device is blacklisted (Device is using VE.Smart).`

naugehyde, 2024-10-24:
VE. Smart is not supported for the moment. Working on a fix as I write
this. In the mean time, you won't get reliable results until you turn off
ve.smart on your device(s)

Peter-Petrik, 2024-10-25:
Thanks for the prompt response. I updated to the just-released v1.1.0-beta.2.1.4 and no longer see the VE.Smart error. 

Values appear to be inline with the Victron app, with the exception of "external device load" which currently shows 1 for all of my chargers, even though there is nothing connected to those outputs.

naugehyde, 2024-10-25:
Re: external device load. Sounds like a parsing error.

I can only test the plugin parsing with the charger's encryption key and encrypted data.

To capture encrypted data run from a command-line:

```bluetoothctl scan on >> petrik_bt_scan.log```

let it run for about 2 minutes then hit ctrl-c to stop it. finally, dm me the log file and encryption key. a screen cap of the victron connect app's instant readout will also be helpful so send that too please. or just let me know what the app is reporting for "load current."

thank you!

Peter-Petrik, 2024-10-25:
Got the file and requested info - what's the best way to DM you or email you the information?

naugehyde, 2024-10-25:
I'm on discord, username is naugehyde.

Or you can email me oh.that.andy@gmail.com

closed 2024-10-26 by naugehyde