naugehyde/bt-sensors-plugin-sk #5: Add Govee sensors

Peter-Petrik, 2024-10-24:
Any possibility of adding Govee sensors to this versatile plugin?

I have the GVH5100 available for testing. The plugin recognizes them, but only allows to enter "Name of sensor" and "Signal strength in db".

The [@signalk-govee-plugin](https://www.npmjs.com/package/signalk-govee-plugin) has the ability to add a very specific Govee device range, and only one per SignalK instance, which is very limiting.

The flexibility of bt-sensors-plugin-sk to add variety of devices and multiple instances of the same device is much more flexible.

naugehyde, 2024-10-25:
I looked into this. I could pretty easily port the existing govee plugin over to mine. But I'm missing identifying characteristics of the Govee product line. If you could send me a BT capture with govee data I could add minimally functional govee support very quickly. More comprehensive support will have to wait until I've get better docs, taken the time to read and understand the existing govee code that's out there.

naugehyde, 2024-10-25:
In addition to a capture log, device info of the scanned govee device would be useful:

To get that, turn your scanner on, determine the govee device(s) mac address by example, AB:CD:EF:01:02:03, and then execute:

```bluetoothctl info <mac address>```

if the scanner has located the device you'll get data like:
[image]

send that to me as well.

thanks

Peter-Petrik, 2024-10-25:
Thank you! I emailed you the log with 3 Govee devices connected, then followed up with the additional info.

closed 2024-10-26 by naugehyde