naugehyde/bt-sensors-plugin-sk #7: Xiaomi LYWSD03MMC

mattnj1975, 2024-11-19:
Hi,
I am trying to connect 4 of these Xiaomi LYWSD03MMC, they show up in the plugin, but I can only add 1 (just get errors if I try and add any more, and it only shows me the signal strength in SignalK...
Do i need to flash them with the custom firmware and/or get the encryption key?
I have tried to connect using the tool:-

https://atc1441.github.io/TelinkFlasher.html

But it doesnt seem to get the MiBindKey anymore....

Will this work with LYWSD03MMC out of the box?

Thanks

mattnj1975, 2024-11-19:
Nov 19 12:45:01 2024-11-19T12:45:01.319Z bt-sensors-plugin-sk Waiting on A4:C1:38:87:7A:D2
Nov 19 12:45:02 2024-11-19T12:45:02.323Z bt-sensors-plugin-sk Found A4:C1:38:87:7A:D2
Nov 19 12:45:02 2024-11-19T12:45:02.352Z bt-sensors-plugin-sk Sensor at A4:C1:38:87:7A:D2 unavailable. Reason: TypeError [ERR_INVALID_ARG_TYPE]: The first argument must be of type string or an instance of Buffer, ArrayBuffer, or Array or an Array-like Object. Received undefined
Nov 19 12:45:02 2024-11-19T12:45:02.353Z bt-sensors-plugin-sk TypeError [ERR_INVALID_ARG_TYPE]: The first argument must be of type string or an instance of Buffer, ArrayBuffer, or Array or an Array-like Object. Received undefined at new NodeError (node:internal/errors:405:5) at Function.from (node:buffer:333:9) at XiaomiMiBeacon.decryptV4and5 (/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/XiaomiMiBeacon.js:154:70) at XiaomiMiBeacon.propertiesChanged (/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/XiaomiMiBeacon.js:170:24) at XiaomiMiBeacon.listen (/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:573:14) at /home/pi/.signalk/node_modules/bt-sensors-plugin-sk/index.js:363:57 at process.processTicksAndRejections (node:internal/process/task_queues:95:5)

mattnj1975, 2024-11-19:
i have actually resolved this by flashing all the devices with custom firmware.

naugehyde, 2024-11-19:
Great! Flashing with ATC (when you can) is the preferred solution. 

The error occurred because the encryption key config variable wasn't set. Which reminds me that I need to add support for required config fields with better runtime checking in general. So, thank you!

In answer to your question, theoretically yes, the plugin supports the Xiaomi out of the box BUT (big but) only with GATT connections which have proven unreliable on some hardware (see warning).

Theoretically, you should have been able to connect all four via GATT connections and get data. Which could have caused other problems with other sensors -- I don't know. I've only tested with one Xiaomi device. And when connecting via GATT it'd occasionally blow up the entire Bluetooth stack.

seamaster101, 2024-12-16:
Hi i did install your plug-in which is so great if it starts working for me!
I try to configure but had no luck...
what do I enter in the in the SignalK paths? they are empty for now and it appear that i can save the plugin settings, but i don't find the data
[image]

naugehyde, 2024-12-16:
Howdy.

Any valid SignalK path will do. It's up to you.

Typically for a temp sensor you'll map the values to environment paths like

environment.indoor.temperature &
environment.indoor.humidity

For the other paths you typically use sensor paths like

sensors.salontempsensor.batterystrength
sensors.salontempsensor.batteryvoltage
sensors.salontempsensor.RSSI

The RSSI refers to the device's radio signal strength

After that you'll be able to display captured values in Kip or other SK
client.

seamaster101, 2024-12-16:
I have custo ATC firmware, but it appears to be missing encryption  key:
`ec 16 01:13:49 2024-12-16T01:13:49.369Z bt-sensors-plugin-sk Error: Unable to get service data for Wireless temp Salon (A4:C1:38:C4:A9:E1 RSSI: -64 db / 62%) ▂ ▄ ▆ at ATC.propertiesChanged (/data/conf/signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/ATC.js:56:19) at BusHelper.<anonymous> (/data/conf/signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:384:26) at BusHelper.emit (node:events:517:28) at ProxyInterface.<anonymous> (/data/conf/signalk/node_modules/node-ble/src/BusHelper.js:41:16) at ProxyInterface.emit (node:events:517:28) at ProxyListener.fn (/data/conf/signalk/node_modules/dbus-next/lib/client/proxy-interface.js:19:18) at EventEmitter.emit (node:events:529:35) at handleMessage (/data/conf/signalk/node_modules/dbus-next/lib/bus.js:122:23) at EventEmitter.<anonymous> (/data/conf/signalk/node_modules/dbus-next/lib/bus.js:151:9) at EventEmitter.emit (node:events:517:28) at /data/conf/signalk/node_modules/dbus-next/lib/connection.js:132:14 at USocket.<anonymous> (/data/conf/signalk/node_modules/dbus-next/lib/message.js:65:9) at USocket.emit (node:events:517:28) at emitReadable_ (node:internal/streams/readable:633:12) at process.processTicksAndRejections (node:internal/process/task_queues:81:21)
Dec 16 01:13:51 2024-12-16T01:13:51.219Z bt-sensors-plugin-sk Error occured on Wireless temperature at A4:C1:38:C4:A9:E1: Wireless temperature at A4:C1:38:C4:A9:E1 requires an encryption key.
Dec 16 01:13:51 2024-12-16T01:13:51.220Z bt-sensors-plugin-sk Error: Wireless temperature at A4:C1:38:C4:A9:E1 requires an encryption key. at XiaomiMiBeacon.propertiesChanged `

naugehyde, 2024-12-16:
Bizarre.

Can you dm me your config file?

it's in the ~/.signalk/plugin-config-data directory

seamaster101, 2024-12-16:
I have Victron Ekrano GX with Venus OS large image which contains SignalK. |I cant find where the files for signalK are. do you know?

naugehyde, 2024-12-16:
Sorry I don't know.

But I'm guessing you have two active configs for the same Xiaomi device.
You need to delete one. Can't figure out what else could be the problem.

seamaster101, 2024-12-16:
what exactly is the file name you are interested in?

naugehyde, 2024-12-16:
bt-sensors-plugin-sk.json

seamaster101, 2024-12-16:
`{
  "configuration": {
    "discoveryTimeout": 30,
    "discoveryInterval": 10,
    "peripherals": [
      {
        "active": false,
        "mac_address": "A4:C1:38:C4:A9:E1",
        "discoveryTimeout": 30,
        "params": {
          "parser": "ATC-LE",
          "name": "Wireless temp Salon"
        },
        "paths": {
          "RSSI": "environment.salon.temp1.signalstregth",
          "voltage": "environment.salon.temp1.battery_lvltage",
          "temp": "environment.salon.temp1.temp",
          "humidity": "environment.salon.temp1.humidity"
        }
      }
    ]
  },
  "enabled": false,
  "enableLogging": false,
  "enableDebug": false
}`

naugehyde, 2024-12-16:
I'm at a total loss to explain the error you're getting.

Try restarting the GX and/or Signal K and let me know what happens.

seamaster101, 2024-12-16:
do you have anydesk?

naugehyde, 2024-12-16:
just installed it

whats your address?

seamaster101, 2024-12-16:
1 818 246 zero one six

seamaster101, 2024-12-16:
the internet connection here is not that great so be patient please

naugehyde, 2024-12-16:
i'm in

seamaster101, 2024-12-16:
Do you want to open voice com?Jordan