naugehyde/bt-sensors-plugin-sk #8: Cannot enter encryption key or signalk paths on Victron Blue smart AC Charger IP65

gdenbuitelaar, 2024-11-19:
Hello,

To start off, this is a really nice plugin. But I can't get it to work with my IP65 charger.

I have been reading in the closed issues and found someone testing a release (gregsyoung) who came to the same conclusion, losing config fields for the IP65 chargers.

Is there a workaround? How I can get the plain text key and the Signalk Paths in the config? 
The log files were talking about a .properties file but I cannot find that anywhere.

Something else, my 24V/13A ip65 charger has 0xA30B as model, I copied "Blue Smart AC Charger IP65" line in VictronConstants.js and added it with that model number, it does come up now as a known device. Maybe you can officially add it.

Thanks
George

naugehyde, 2024-11-19:
Hi George. 

Happy to help. Let's see what we need to do get you up and working.

If you're not seeing the encryption key config parameter and SK path configs, something is wrong-- probably in my code.

To figure this out, please confirm the plugin release number on the installed plugin page (SK->Appstore->Installed).

If it's not up to date. Please update, restart and try again. 

If you're still not seeing an encryption key parameter, please execute the following from a command prompt

`journalctl  -u signalk.service -b | gzip -c>> gdenbuitelaar-sk.log.gz`

then DM me or attach to this thread the gdenbuitelaar-sk.log.gz file.

Also, if there are screenshots that will help debug, please post.

Thank you!

gdenbuitelaar, 2024-11-20:
Hi, 

The plugin release number is v1.1.0-beta.2.2.0 (2024-11-16). There were 7 other plugins not up to date so I updated everyting I could find :) but that did not help.

I pasted the log in and added some screenshots that probably wont help.

[gdenbuitelaar-sk.log.gz](https://github.com/user-attachments/files/17830364/gdenbuitelaar-sk.log.gz)
[image]
[image]

naugehyde, 2024-11-20:
Before sending anything more try the latest Beta 2.2.0.1

It should be in the appstore.

gdenbuitelaar, 2024-11-21:
Hi, That was quick :) I deleted the plugin then reinstalled it. Then suddenly after some waiting I got the fields and entered the enc key. But still no luck , I turend on debug logging and found, the value of "offset" is out of range. Is 5 should be between =>0 and <= 2.

Strange here is a screenshot.

[image]

naugehyde, 2024-11-21:
Can you check to see that the Charger's firmware is up to date?

naugehyde, 2024-11-21:
If it's not, upgrade to the latest firmware and try again.

gdenbuitelaar, 2024-11-21:
Hi, 

The firmware is v3.25, it sais that this is the latest version. I switched it from lithium to normal batery now I get even less info :) the ip65's look a bit alike on the outside but you never know what they did on the inside :)

I'll run some more tests with this one this evening and tomorrow removing everything again and reinstalling the plugin.

Also this weekend I am taking this raspberry pi and screen plotter built that runs the ksensor, to the boat where I got a blue solar a smartshunt and an ip65 12V. I am most interested if I can get the smartshunt to read then the smartsolar. The smartshunt is on the starter battery and the smartsolar on the house batery. The AC ip65 chargers are just extra and this 24 V 13A ip65 I got at home so I used that for testing initially.

I'll keep you up to date with testing.

Thanks so far :)
George

naugehyde, 2024-11-21:
Ok, let me know how the rest of the testing goes.

The error you're seeing is a result of a packet that's only 3 bytes long. Encrypted instant readout packets are, I believe, 8 bytes plus payload. 16 bytes in total, I think. 

I tested on an ip65 I found on a boat on my dock and had to upgrade the owner's firmware before I'd get readout data that I could decrypt and display. Maybe the 0xA30B models are an outlier?

naugehyde, 2024-11-21:
There's a new Victron Connect version available as of today. 

From the readme (emphasis mine):

> 
> Phoenix & Blue Smart Chargers firmware version v3.61:
> 
> Applies to all 230V- as well as all 120-240V Phoenix Smart Charger models
> 
> **Applies to all IP22, IP65 and IP67 Blue Smart Chargers produced after a hardware change made in production near the end of 2019. _Chargers of the earlier generation can be updated to v3.25 and no further - VictronConnect App will then show a “this the latest version” note._ Chargers of more recent hardware generations can be updated to v3.61. There are no plans to make firmware updates available for the earlier generation of chargers.**
> 
> Firmware v3.61 change log:
> Add Instant Readout
> Fix low temperature Cut-Off (IP22 and IP43)
> Add synchronized charging to IP43 230V
> Time for fixed absorption can now be set to 0 on all chargers (disabling absorption)
> Repeated absorption can now be disabled on all chargers (requires Victron Connect update)
> Adaptive absorption time can now be set to 30 minutes on all chargers.
> Tail current can be configured on all chargers
> Add 30 seconds bluetooth enable option to BSC IP65 6-12V/1.1A since it has no mode button
> Update PUK entry procedure when 30 seconds BT is configured

gdenbuitelaar, 2024-11-22:
Hello,

SUCCESSS!!!!!

In the app store I updated the victron connect app and then it found the 3.65 update for the charger it is from 2022 the first two numbers in the serial number are 22.

It is now displaying too in opencpn super awesome!!!!! Really happy with it.

Thanks you for the super fast support!!! I will let you know how the other devices go on board. And if there is anything I can do to help please let me know.

[image]

[image]

naugehyde, 2024-11-22:
Happy to hear!

And man is the air dry in the Netherlands!

gdenbuitelaar, 2024-11-22:
Hi, yes that is a BME280 sensor attached to the PICAN-M i2c port on the board I was testing it putting it in a sealed bag with silica and a normal humid scale that is probably why it is so low :) it is actually 44 in the house and raining outside :) 
And yes I live in Netherlands Rotterdam close to river the Nieuwe Maas :)

naugehyde, 2024-11-22:
Love the Pican-M. Mine has the SMPS and don't know how I'd live without it.

Do you get good readings from the BME280 when not in a sealed bag? My rpi4b
and Pican run hot enough that I'm sure they'd dry out their immediate
environment and any on board temp or humidity sensor would report spurious
values.

FYI there are plenty of BT temp and humidity sensors out there. Several
supported by my plugin. I can recommend the Ruuvi Tag which is a little
pricey compared to the standard cheap-o Chinese sensor but pretty darn
reliable.

gdenbuitelaar, 2024-11-30:
Hello, The readings of the BME280 give simular values as the normal cheap chines ones in the end. I hooked up the BME280 in the enclosure where my raspberry is to see the env temp next to cpu temp to see if any additional cooling is needed, never mind the humidity or pressure:). I'll get some ruuvi tags for the engine compartment and in cabin and outside environment,that will be a good addition to the sensors in general :)

I also took my setup to the boat today and tested the Victron smartshunt, blue solar and  ip65 12V battery charger. I could upgrade the first 2.

-The Smartshunt I bought in Januari of this year I can attach and put the encryption key in, but it reads strange values jumping around random numbers it seems on all readings.
I can turn on GATT on the smartshunt and the plugin so I did but then it says I need to pair it to signalk. I thought putting in the enc key was kind of pairing it . How do I pair it to signalk and would that even solve the issue?
 
-The victron blue solar 15/75 worked without problems !!!!!!
-The charger has simular problems as the charger at home before upgrade, but this is a 2018 charger so I cannot further upgrade it, never mind I need to get a new one anyway and can use this one elsewhere.

naugehyde, 2024-11-30:
Hi George. The SmartShunt may have some parsing errors  (unlikely) or it may not be connected correctly to the batteries. Are you getting good readings from VictronConnect? 

If you are getting good readings on VC it's possible that the encryptionkey was not set correctly in the config. Double check that first. 

Or it's possible that your SmartShunt is a different model than the one I developed for. Send a screencap of the VictronConnect screen of the SmartShunt just to be sure.

Before we go to the GATT connection I'd like to test the above.

To pair the smartshunt: 

```bluetoothctl
scan on
pair <mac address of shunt>```

You should get a enter PIN code prompt. Enter the PIN that you set in the VC app or 000000 (default PIN code)

gdenbuitelaar, 2024-11-30:
Hi,

I am not at the boat now, but did take my raspberry back home I just read the settings I used and I did made a screenshot at the boat earlier today of the encryption key on my phone. I just compared them and...

Stupid me, I made a typo in the encryption key!!!!

I'll try again tomorrow.

Thanks again

naugehyde, 2024-11-30:
I don't know if there's a way to verify the encryption key. But I'll look into it!

naugehyde, 2024-11-30:
There is a way but it's not 100% reliable. I'll put it in the next release nevertheless.

gdenbuitelaar, 2024-11-30:
Nice, In the form of a warning that should be okay if it is not 100% reliable.
I am just curious, do you test on a predefined always the same value in the data? Like they might have the string "Victron"or something always in the data so if the password is correct you'll find that.

naugehyde, 2024-11-30:
The test is not that great. It's just the first two bytes of the encryption key embedded in the encrypted packet. Tested a few common transcription errors and its not looking too useful. I'll reach out to Victron and see if better awaits us.

Usually its the BT mac address that's provided in the decrypted packet as a check but that's 12 bytes that would otherwise be used for data.

seamaster101, 2024-12-16:
@naugehyde thank you for helping me earlier today with my setup of the temperature sensor!
With regards to my Victron Blue Smart IP22 charger not available on the list for selection, I think I figured out what is going on. My Cerbo GX is far way from some of my Bluetooth devices, therefore I have USB extension cable with external Bluetooth adapter attached to the Cerbo GX. This makes two Bluetooth adapters available to the system. I'm speculating that the your plug-in is using the Bluetooth that is built in to the Cerbo GX rather the external one, thus the Victron Blue Smart Charger is outside of the range and it does not become available for the selection in the plugin configuration. 
Is there a way to figure this out or select which Bluetooth adapter the plugin should use?

naugehyde, 2024-12-16:
there isn't one currently but I can add one.

put up a feature request in github and i'll get on it.