naugehyde/bt-sensors-plugin-sk #9: new feature request

seamaster101, 2024-12-16:
My Victron Blue Smart IP22 charger is not available on the list for selection. I think I figured out what is going on. My Cerbo GX is far way from some of my Bluetooth devices, therefore I have USB extension cable with external Bluetooth adapter attached to the Cerbo GX. This makes two Bluetooth adapters available to the system. I'm speculating that the your plug-in is using the Bluetooth that is built in to the Cerbo GX rather the external one, thus the Victron Blue Smart Charger is outside of the range and it does not become available for the selection in the plugin configuration.
Can you please implement selection of which Bluetooth adapter to be used?

naugehyde, 2024-12-17:
There's a new beta in the app store as of two minutes ago.

You can select your BT adapter in the config now. If there's only one adapter the list is read only.

seamaster101, 2024-12-17:
Thank you very much! I see you made it as general selection. Was just thinking out loud: what if one adapter is covering the forward portion of the boat and another adapter is powering the aft portion of the boat? the range of the Bluetooth devices is poor, so that could be very likely scenario. wouldn't capable to select adapter be better on per device bases?

naugehyde, 2024-12-17:
I'll definitely consider that for future releases.

seamaster101, 2024-12-18:
Hi,
I tested today and while I was able to change adapter, the new adapter doesn't read available sensors. I'm available if you have time to debug

naugehyde, 2024-12-18:
Strange.

Did you restart the plugin and/or signalk?

seamaster101, 2024-12-18:
it happened again here is the log:
`Dec 18 00:52:00 bad MMSI: atons.urn:mrn:imo:mmsi:3160171
Dec 18 00:52:09 2024-12-18T00:52:09.707Z bt-sensors-plugin-sk Unable to communicate with device A4:C1:38:C4:A9:E1 Reason: operation timed out
Dec 18 00:52:09 2024-12-18T00:52:09.714Z bt-sensors-plugin-sk Error: operation timed out at Timeout.<anonymous> (/data/conf/signalk/node_modules/node-ble/src/Adapter.js:158:16) at listOnTimeout (node:internal/timers:569:17) at process.processTimers (node:internal/timers:512:7)
Dec 18 00:52:09 2024-12-18T00:52:09.718Z bt-sensors-plugin-sk Sensor at A4:C1:38:C4:A9:E1 unavailable. Reason: operation timed out
Dec 18 00:52:09 2024-12-18T00:52:09.721Z bt-sensors-plugin-sk operation timed out
Dec 18 00:52:10 2024-12-18T00:52:10.869Z bt-sensors-plugin-sk Unable to communicate with device DA:37:91:24:E3:91 Reason: operation timed out
Dec 18 00:52:10 2024-12-18T00:52:10.876Z bt-sensors-plugin-sk Error: operation timed out at Timeout.<anonymous> (/data/conf/signalk/node_modules/node-ble/src/Adapter.js:158:16) at listOnTimeout (node:internal/timers:569:17) at process.processTimers (node:internal/timers:512:7)
Dec 18 00:52:10 2024-12-18T00:52:10.879Z bt-sensors-plugin-sk Sensor at DA:37:91:24:E3:91 unavailable. Reason: operation timed out
Dec 18 00:52:10 2024-12-18T00:52:10.881Z bt-sensors-plugin-sk operation timed out
Dec 18 00:52:31 GET /skServer/plugins 200 78.860 ms - -
Dec 18 00:52:36 GET /skServer/plugins 304 48.579 ms - -
Dec 18 00:53:09 GET /skServer/plugins 304 47.909 ms - -
Dec 18 00:53:17 2024-12-18T00:53:17.871Z bt-sensors-plugin-sk Stopping plugin
Dec 18 00:53:17 POST /skServer/plugins/bt-sensors-plugin-sk/config 200 15.415 ms - 53
Dec 18 00:53:18 2024-12-18T00:53:18.223Z bt-sensors-plugin-sk Error stopping scan: No discovery started
Dec 18 00:53:18 2024-12-18T00:53:18.224Z bt-sensors-plugin-sk BT Sensors plugin stopped
Dec 18 00:53:21 GET /skServer/logfiles/ 304 18.175 ms - -
Dec 18 00:53:22 GET /skServer/debugKeys 200 7.850 ms - 
`

naugehyde, 2024-12-18:
I'll anydesk back in.

Please keep the message dialog closed (i see it and it gets in the way).

I'll let you know if i find anything.

eburi, 2025-05-19:
Hi,
I have the same issue, my sensors are so far apart on the boat that the Bluetooth Adapter is not picking up the signal. 
I'm thinking about adding an extra adapter to extend the reach.

But if I understand correctly, I can only use one adapter. Is that correct?
It would be great, if it would scan on all available adapters becaus I think that I would have to place several Bluetooth adapters over the boat to reach all sensors. I don't think I can place one adapter in a place that would be in reach of all sensors on board. 

Ideally I don't have to put additional Raspberry Pis with separate SignalK installations with each Bluetooth adapter.

Please let me know if anything is planned, or if I could maybe help change the code accordingly.

naugehyde, 2025-05-19:
> I've had good results with a more powerful USB BT adapter installed on my
> boat's pi. (TP-Link UB500 Plus) which as of its latest firmware is pretty
> solid.
>
> You may have to unblock the UB500 with rfkill but otherwise it's a fairly
> easy install.
>
> A hint: try and keep your server and adapter away from RF interference
> (2.4Ghz devices and HDMI can cause interference for example).
>
> Also, what devices are you connecting to? Connected devices can be finicky
> in noisy environments -- especially with the RPi's onboard bluetooth
> adapter.

Also, I'm looking into supporting multiple bluetooth adapters in future
versions but that won't solve the adapter proximity problem (unless you run
USB extensions to USB adapters I suppose).