naugehyde/bt-sensors-plugin-sk #10: New sensor needed

hugorobi, 2024-12-18:
Hi !

Nice work for this plug-in !

I would like to help developping on OpenPlotter and all tools included with. Unfortunatly, I have obsolutely NO knowledge on object programming, so I have to ask for new feature. The good news is that I think I've found all the information needed to add my Temperature/Hyg sensor.

I have few units of this Temp/Hyg sensors :
https://www.switch-bot.com/products/switchbot-indoor-outdoor-thermo-hygrometer

I've found the documentation here : 
https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/blob/latest/devicetypes/meter.md#outdoor-temperaturehumidity-sensor

I have my new Openplotter project and SwitchBot Temp sensor with me, so I can have test done "on-demand".

Regards,

Hugo

naugehyde, 2024-12-18:
Looks like an easy one to implement. And a device I'd like to support.

Let me see what I can do.

I'll let you know when a version is ready for testing.

naugehyde, 2024-12-18:
In the meantime, since I don't have a device to work with, if you could DM me a bluetooth capture.

From an OpenPlotter command line, execute the following:

```
bluetoothctl scan on >> hugorobi_bt.cap 
```

After about 3 minutes hit ctrl-c to stop the capture. Then execute:

```
bluetoothctl info [mac address of switch bot goes here] > hugorobi_bt.cap
```

hugorobi, 2024-12-18:
Hi,

I'm pretty new here. How to DM ???

naugehyde, 2024-12-18:
github removed the private message feature so just reply to this comment with the file attached

hugorobi, 2024-12-18:
Hi,

Here is the file.

Thanks !
[hugorobi_bt.cap.gz](https://github.com/user-attachments/files/18187544/hugorobi_bt.cap.gz)

naugehyde, 2024-12-18:
One more thing. Forgot to ask. 

Can you repeat the cap and send but with the SwitchBot readings (temp humidity and battery power) 
from whatever app you use?

Thanks!

hugorobi, 2024-12-18:
Hi,

Capture done on my Raspberry. While capturing, launched SwitchBot App on my phone to have data sync.

Thanks,

Hugo

naugehyde, 2024-12-19:
Please attach data!

hugorobi, 2024-12-19:
Hi !

The file was attached to my email, but did not appear here.

This reply is from Github website.

Thanks
[hugorobi_bt.cap.gz](https://github.com/user-attachments/files/18190737/hugorobi_bt.cap.gz)

naugehyde, 2024-12-19:
Hi Hugo,

There's a new class called SwitchBotTH for you to test. Easiest way to install is to

1) go here:

https://github.com/naugehyde/bt-sensors-plugin-sk/blob/1.1.0/sensor_classes/SwitchBotTH.js 

2) download the raw file by clicking on the down arrow

[image]

3) Copy the file into the plugin's source directory

```
cp \path\to\SwitchBotTH.js ~\.signalk\node_modules\bt-sensors-plugin-sk\sensor_classes
```

4) Restart Signalk 

You should see your device in the config now and be able to map SK paths to its values.

I've only done minimal testing so please let me know how it goes. If it's working, I'll put the SwitchBotTH class in the next release.

hugorobi, 2024-12-19:
Hi,

Good news... I'm able to have readed values added in SignalK browser list.

I will do more extensive test tomorrow.

Many thanks

Hugo

-------- Message d'origine --------

https://github.com/naugehyde/bt-sensors-plugin-sk/blob/1.1.0/sensor_classes/SwitchBotTH.js

  1.  download the raw file by clicking on the down arrow

Screenshot.2024-12-18.at.7.05.40.PM.png (view on web)<https://github.com/user-attachments/assets/f2080b20-a580-4ead-8a74-39886b170eb3>

  1.  Copy the file into the plugin's source directory

  1.  Restart Signalk

hugorobi, 2024-12-19:
Hi,

I have only 1 device and it work fine. No issue. Thanks.

I have an inverter/charger (Xantrex) that I would also like to have connectivity with SignalK, but can't find document about Bluetooth specifications.

Do you used the PDF documentation or were you able to create the new class only with capture provided ?

Thanks again !

naugehyde, 2024-12-19:
Go ahead and open up a new issue/feature request for the Xantrex. I'll look into it. FYI There's a large number of BMS and chargers out there with undocumented BT data streams that I'm looking into currently.

The PDF docs were of primary help. The captured data was essential for testing.

closed 2024-12-19 by hugorobi

naugehyde, 2024-12-20:
@hugorobi  check out: https://github.com/naugehyde/bt-sensors-plugin-sk/issues/11#issue-2753268490

naugehyde, 2025-03-17:
Hi Hugo,

I'll need remote command line access to your Signalk server if there's any chance I can get Xantrex support into the plugin. 

Would that be possible?

Best,

Andy