naugehyde/bt-sensors-plugin-sk #11: Xantrex support

naugehyde, 2024-12-20:
Add support for Xantrex devices. 

Xantrex inverters and chargers etc connect by wire to a panel that emits Bluetooth (https://xantrex.com/products/accessories/freedom-x-bluetooth-remote-panel/).

NOTE: The BT protocol is not documented and, to all evidence, not yet been reverse engineered.

naugehyde, 2025-03-17:
Hi @hugorobi,

I'll need remote command line access to your Signalk server if there's any chance I can get Xantrex support into the plugin.

Would that be possible?

Best,

Andy

hugorobi, 2025-03-20:
Hi @naugehyde,

My Xantrex is on boat and my Raspberry is with me at home. There is still snow here.

Can you give me the command line that you need to have the output ?

I will try to send you the output as soon as I can (few weeks, mid april).

Thank you.

Hugo