naugehyde/bt-sensors-plugin-sk #12: Support for Smart Batteries with JBD BMS

Techstyleuk, 2025-01-02:
I will be receiving a new 300Ah smart battery with a Bluetooth JBD BMS (https://www.docanpower.com/index.php?route=product/product/download&product_id=556&download_id=86)

There appears to be many apps that support JBD and some Node based projects.  As such I asked Perplexity to create a Sensor Class and this is what was generated.  I don't have the battery yet to test, and I don't know if this is any help for you?

[JBDBMS.js.txt](https://github.com/user-attachments/files/18293482/JBDBMS.js.txt)

are there other updates needed?  for instance a reference in index of sensor classes?

Techstyleuk, 2025-01-02:
I haven't asked them about the App yet, but as they do not cover it on the Website, I think they may rely on the many generic apps out there for this BMS.  I will ask them once they ship my battery - I have been bombarding them with questions for a while and I may be trying their patience!!

I did find a bunch of project using JdbBms.py (https://github.com/rakhmaevao/JbdBms) or https://github.com/tgalarneau/bms

naugehyde, 2025-01-02:
I found this
https://github.com/fl4p/batmon-ha/blob/master/bmslib/models/jbd.py

which I think is the model you're aiming for. If so, let me know.

Techstyleuk, 2025-01-02:
Some really good references in there.

As soon as I get the Battery (around 10th January) I will see what info I can pull from it with one of the apps and perhaps the Pi.  I will have the Pi in my basement with the battery over winter so access will be easier than it was for the Renogy, if you need it.

naugehyde, 2025-01-02:
Great. Send Bluetooth captures if you can. The protocol looks similar to
the Renogy (transmit a request command on the GATT  TX service then wait on
a response on the RX service). Not the best way to do it by far but we
should be able to get something working. I may need to ssh in to your
server to test and finalize the code. Does your server have an Internet
gateway?

Techstyleuk, 2025-01-02:
Yes, it will be on my home network all winter, so it will be the same Pi you logged into last time using Zerotier.

Techstyleuk, 2025-01-18:
I using the project - https://github.com/tgalarneau/bms, I had to create a virtual environment and then I installed 'bluepy' then ran the commands below and got the following output

```
(bpenv) pi@openplotter:~/bms $ python jbdbms-4-socket-2temps.py -b A5:C2:37:2C:B3:6F -i 10 -m jbdbms
attempting to connect
connected  A5:C2:37:2C:B3:6F
Error creating socket: [Errno 2] No such file or directory
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.41,19.63,263.24,166,300,1
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,cell1,cell2,cell3,cell4
jbdbms,3354,3354,3356,3354
meter,cellmin,cellmax,delta
jbdbms,3354,3356,2
```

To decode this we have:
jbdbms,13.41,19.63,263.24,166,300,1
jbdbms,<Battery_Voltage>,<Battery_current_+is_charge>,<Power>,<remaining_capacity>,<Total_capacity>,<charge cycles>

jbdbms,3354,3354,3356,3354
jbdbms,<Cell1_Voltage>, <Cell2_Voltage>, <Cell3_Voltage>, <Cell4_Voltage>

jbdbms,3354,3356,2
jbdbms,<Min_Cell_Voltage>,<Max_Cell_Voltage>, <Max_Cell_Voltage_minus_Min_Cell_Voltage>

Techstyleuk, 2025-01-18:
Here are a couple of Screen shots of the App:

[image]

[image]

naugehyde, 2025-01-20:
Thank you. 

I'm going to need a bluetooth capture. 

run 

`bluetoothctl scan on >> jason_bt_cap.log`

from a command line for about 3 minutes then dm me the log file.

Thanks!

Techstyleuk, 2025-01-20:
Data sent via Discord chat

naugehyde, 2025-01-22:
Can you DM the results of:

`bluetoothctl info A5:C2:37:2C:B3:6F `

Thanks.

Techstyleuk, 2025-01-23:
First I ran 

`bluetoothctl scan on`

and got

```
pi@openplotter:~ $ bluetoothctl scan on
Discovery started
[CHG] Controller D8:3A:DD:C1:94:30 Discovering: yes
[NEW] Device A5:C2:37:2C:B3:6F DP04S007L4S200A
[NEW] Device A4:C1:38:B5:55:F2 LYWSD03MMC
[NEW] Device A4:C1:38:F2:49:3F LYWSD03MMC
[NEW] Device A4:C1:38:43:2B:19 LYWSD03MMC
[NEW] Device 40:B8:9A:1A:68:54 Living Room
```
and then:

`pi@openplotter:~ $ bluetoothctl info A5:C2:37:2C:B3:6F
Device A5:C2:37:2C:B3:6F not available
`
Not sure why it is unavailable, when its not

naugehyde, 2025-01-24:
Try letting the scanner run for longer-- until it finds the battery.

Techstyleuk, 2025-01-25:
Ran for a while:

`^Cpi@openplotter:~ $ bluetoothctl info A5:C2:37:2C:B3:6F
Device A5:C2:37:2C:B3:6F (public)
        Name: DP04S007L4S200A
        Alias: DP04S007L4S200A
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        ManufacturerData Key: 0xb36f
        ManufacturerData Value:
  2c 37 c2 a5                                      ,7..
        RSSI: -59
`

Techstyleuk, 2025-01-25:
[image]

naugehyde, 2025-01-26:
That's very helpful.

Looks like the only identifying characteristic is the Name/Alias. The Manufacturer Data is just the device mac address in reverse. 

From what I can see the model name is DP04S007. 

I would like to reliably support other BMS from JBD (looks like they have a bunch of different models) but will start with just this one.

closed 2025-04-13 by naugehyde