naugehyde/bt-sensors-plugin-sk #14: Negative temperatures not converting correctly on Inkbird sensor

195quebecave, 2025-02-05:
I have an INKBIRD device I put in the freezer. It appears that when there is a negative value (in Celius) on the INKBIRD app it does not convert the Kelvin value in SignalK correctly. In the below example it shows as 921.16 Kelvin which is not correct.

environment.inside.freezer.temperature | 921.16 | K | 02/05 07:30:39 | bt-sensors-plugin-sk

naugehyde, 2025-02-05:
If it's the TH-2, please go to the command line and send the result of:

`bluetoothctl info <mac address of inkbird device>`

If you get an error message turn scanning on, wait a minute and retry.

I think I know what the problem is but I don't have a good way to test subzero temps on the device.

naugehyde, 2025-02-05:
There's a new version available in the SK app store that attempts to address the issue of negative C Inkbird values. I am unable to do a live test but if you could install the update and test it with your Inkbird and report any errors or issues here, I'd be much obliged.

195quebecave, 2025-02-06:
InkBird - Bluetooth Hygrometer Thermometer IBS-TH2

I will give the new version a try and report back.

195quebecave, 2025-02-06:
The updated release now shows 259.13 Kelvin when the InkBird temperature reads -14.26 Celsius.

environment.inside.freezer.temperature | 259.13 | K | 02/06 08:45:11 | bt-sensors-plugin-sk

Based on this test I would say the issue has been resolved. Thank You.

closed 2025-02-06 by 195quebecave