naugehyde/bt-sensors-plugin-sk #18: Plugin does not recover from interface being turned off and on

sebastianhaas, 2025-02-25:
While running signalk with bt-sensors-plugin-sk installed and active, turning the Bluetooth interface off and then on again results in the following crash, which the plugin does apparently not recover from:

```
  bt-sensors-plugin-sk Found 65:51:BB:ED:23:E5 +0ms
  bt-sensors-plugin-sk Found 71:24:8D:A7:DE:EF +2ms
  bt-sensors-plugin-sk instantiated 41:F8:E6:4C:6B:1C +5ms
  bt-sensors-plugin-sk Device at 41:F8:E6:4C:6B:1C unavailable. Reason: Device is blacklisted (Randomized MAC address (Apple)). +1ms
  bt-sensors-plugin-sk instantiated 4F:48:9F:65:7A:8C +0ms
  bt-sensors-plugin-sk Device at 4F:48:9F:65:7A:8C unavailable. Reason: Device is blacklisted (Randomized MAC address (Apple)). +0ms
Error: interface not found in proxy object: org.bluez.Adapter1
    at ProxyObject.getInterface (/home/sebastian/.signalk/node_modules/dbus-next/lib/client/proxy-object.js:81:13)
    at BusHelper._prepare (/home/sebastian/.signalk/node_modules/node-ble/src/BusHelper.js:32:42)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
    at async BusHelper.children (/home/sebastian/.signalk/node_modules/node-ble/src/BusHelper.js:91:5)
    at async Adapter.devices (/home/sebastian/.signalk/node_modules/node-ble/src/Adapter.js:106:21)
Error: interface not found in proxy object: org.bluez.Adapter1
    at ProxyObject.getInterface (/home/sebastian/.signalk/node_modules/dbus-next/lib/client/proxy-object.js:81:13)
    at BusHelper._prepare (/home/sebastian/.signalk/node_modules/node-ble/src/BusHelper.js:32:42)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
    at async BusHelper.children (/home/sebastian/.signalk/node_modules/node-ble/src/BusHelper.js:91:5)
    at async Adapter.devices (/home/sebastian/.signalk/node_modules/node-ble/src/Adapter.js:106:21)
```

naugehyde, 2025-03-13:
Looking into this. Initial investigation suggests that neither Node-ble nor DBus-Next emits an event when the Adapter's state changes. Can you confirm?

naugehyde, 2025-03-14:
Check out and test https://github.com/naugehyde/bt-sensors-plugin-sk/commit/0bf83c0109a786cbf78303f4d48820a14b310fea

Should do the trick for you. 

Tested from bluetoothctl turning Powered off and on for the current adapter.

sebastianhaas, 2025-03-15:
Hi Andy

I gave it a try and it works flawlessly on my system! I can now turn on and off Bluetooth and plugin will recover just fine.

There is one thing I noticed though, and I have never observed this before so there is a good chance it was introduced by your changes:

```
(node:207054) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 {"path":"/org/bluez/hci0","interface":"org.freedesktop.DBus.Properties","member":"PropertiesChanged"} listeners added to [EventEmitter]. MaxListeners is 10. Use emitter.setMaxListeners() to increase limit
```

I ran signalk five times with your changes, and five times without and I have a 100 percent rate of seeting the error with updates and 0 % without them. I know, it is just a warning and yould increase the limit, but I wonder if you can make sense of it. I have only configured 2 sensors in my current setup so even with the additional subscription for the adapter state it shouldn't be 11?

naugehyde, 2025-03-15:
The EventEmitter warning is a known issue in Node-BLE. Getting it on my end as well.

The relevant change I made was asking Adapter object in Node-BLE to send propertyChange events but the warning occurs in subsequent calls to EventEmitter::on().

I'll look into it but it shouldn't cause any runtime memory issues as the warning only comes up once at plugin load. If it does present any serious issues on your end, please let me know.

Keeping issue open for now.

naugehyde, 2025-03-15:
EventEmitter problem should be fixed. 

NB: It's a bug/limitation of Node-BLE that I'm grateful for (wouldn't have gotten this far without it) but would love to replace with a better supported package.

closed 2025-03-25 by naugehyde