naugehyde/bt-sensors-plugin-sk #20: Comprehension problems

bickes-cgn, 2025-02-27:
Hi,
- I installed the Plugin to read Data from my Vitron-solar-charger and my Victron -Shunt.
- I can see the BT adapter in the plugin config
- I can see and select the Victrons in the plugin config.
But I cannot get data from the sensors.
My questions:
- Do I have to pair the rpi and the devices before using the plugin and where do I do that? (the devices do not appear in the rpi-list of pairable devices, but speakers etc. do)
- Where do I get the encryption key from, which is required in the plugin config? At least the log sasy, it is required 
- Do the signalk keys, that can be updated by the plugin, have to be created in advance or will they be created by the plugin?

Below you find an extract from the log!

Thanks for helping to understand that great plugin!

[image]

[image]

`Feb 27 14:47:11 2025-02-27T13:47:11.382Z bt-sensors-plugin-sk Error: Unable to read data from Victron SmartSolar MPPT 75|15 (FB:C3:A7:54:1E:2F RSSI: -85 db / 23%) â : Error: Unable to decrypt: no encryption key set at VictronSolarCharger.propertiesChanged (/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/Victron/VictronSensor.js:100:19) at BusHelper.<anonymous> (/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:384:26) at BusHelper.emit (node:events:514:28) at ProxyInterface.<anonymous> (/home/pi/.signalk/node_modules/node-ble/src/BusHelper.js:41:16) at ProxyInterface.emit (node:events:514:28) at ProxyListener.fn (/home/pi/.signalk/node_modules/dbus-next/lib/client/proxy-interface.js:19:18) at EventEmitter.emit (node:events:514:28) at handleMessage (/home/pi/.signalk/node_modules/dbus-next/lib/bus.js:122:23) at EventEmitter.<anonymous> (/home/pi/.signalk/node_modules/dbus-next/lib/bus.js:151:9) at EventEmitter.emit (node:events:514:28) at /home/pi/.signalk/node_modules/dbus-next/lib/connection.js:132:14 at Socket.<anonymous> (/home/pi/.signalk/node_modules/dbus-next/lib/message.js:65:9) at Socket.emit (node:events:514:28) at emitReadable_ (node:internal/streams/readable:590:12) at process.processTicksAndRejections (node:internal/process/task_queues:81:21)
Feb 27 14:47:11 2025-02-27T13:47:11.980Z`

bickes-cgn, 2025-02-28:
That doesn´t seem to be too complicated. I will try that next week, when I am at the boat.
Thanks so far. I will report if it works (and come back with more questions if not).

bickes-cgn, 2025-03-03:
Thanks for your hints. It worked at the first try. All my three Victron devices send data. 
Once again learned something new.

naugehyde, 2025-03-03:
Great to hear. Closing this issue! Good luck!

closed 2025-03-03 by naugehyde