naugehyde/bt-sensors-plugin-sk #21: New device - Lancol BT battery voltage sensor

lula6894, 2025-03-03:
Thanks for this fantastic Plugin.

Lancol 12V Bluetooth 4.0 Battery Tester Micro-10-C (https://www.lancol.com/product/12v-bluetooth-4-0-battery-tester-micro-10-c/) is a unexpensive device to test battery voltage. Just attach to the positive and negative and transmit in the "Device name" the voltage like: "Lancol 12.08V".

The proposed enhancement is to have Lancol in the BT list of devices (currently the plugin see it) and to be able to assign the voltage to a Signalk variable.

naugehyde, 2025-03-04:
There's no public docs on the device's BT protocol that I could find, so I need you to do the following:

From a command line, execute the following:

`bluetoothctl scan on > lancol_bt.cap `

After about 5 minutes hit ctrl-c to stop the capture. Then execute:

`bluetoothctl info [mac address of device goes here] >> lancol_bt.cap`

Thank you!

lula6894, 2025-03-04:
Thanks for getting care of the improvement so fast!!!

This is the only information I have from the Lancol: the voltage is transmitted after "Lancol " in the BT "Device name" . Look at the picture. 
[image]

Attached is the lancol_bt.cap file after captures

[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -79
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -79
[[0;93mCHG[0m] Device A4:C1:38:E1:20:83 RSSI: -77
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -83
[[0;93mCHG[0m] Device A4:C1:38:DF:2D:37 RSSI: -58
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -81
[[0;93mCHG[0m] Device A4:C1:38:2D:ED:07 RSSI: -75
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -79
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -81
[[0;93mCHG[0m] Device A4:C1:38:E1:20:83 RSSI: -78
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -77
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -81
[[0;93mCHG[0m] Device A4:C1:38:DF:2D:37 RSSI: -59
[[0;93mCHG[0m] Device A4:C1:38:2D:ED:07 RSSI: -73
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -79
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -78
[[0;93mCHG[0m] Device A4:C1:38:2D:ED:07 RSSI: -61
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -79
[[0;93mCHG[0m] Device A4:C1:38:E1:20:83 RSSI: -76
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -82
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -83
[[0;93mCHG[0m] Device A4:C1:38:E1:20:83 RSSI: -76
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -80
[[0;93mCHG[0m] Device A4:C1:38:DF:2D:37 RSSI: -59
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -79
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -78
[[0;93mCHG[0m] Device A4:C1:38:E1:20:83 RSSI: -78
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -84
[[0;93mCHG[0m] Device A4:C1:38:DF:2D:37 RSSI: -54
[[0;93mCHG[0m] Device A4:C1:38:DF:2D:37 ServiceData Key: 0000181a-0000-1000-8000-00805f9b34fb
[[0;93mCHG[0m] Device A4:C1:38:DF:2D:37 ServiceData Value:
  a4 c1 38 df 2d 37 00 a8 3f 2d 0a 4d c2           ..8.-7..?-.M.   
[[0;93mCHG[0m] Device A4:C1:38:2D:ED:07 RSSI: -75
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -79
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -78
[[0;93mCHG[0m] Device A4:C1:38:DF:2D:37 RSSI: -58
[[0;93mCHG[0m] Device A4:C1:38:2D:ED:07 RSSI: -76
[[0;93mCHG[0m] Device C8:FD:19:02:33:96 RSSI: -77
[[0;93mCHG[0m] Device A4:C1:38:DF:2D:37 RSSI: -58
Device C8:FD:19:02:33:96 (public)
	Name: Lancol    12.08V
	Alias: Lancol    12.08V
	Paired: no
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	RSSI: -81

Hope it helps, Thanks again my friend

naugehyde, 2025-03-05:
Hi Lula.

Can you please re-run the capture for longer? Ten minutes or so. 

Also, please attach the capture file by clicking on the paperclip icon below. It looks like: ([image])

Also, run the following code from a command line and send the results, too:

`busctl introspect org.bluez /org/bluez/hci0/dev_C8_FD_19_02_33_96`

Thanks!

lula6894, 2025-03-05:
Sure I do. I run the capture for about 10 min.
Files attached (changed to .jpg)

[image]
[image]

Thanks again

naugehyde, 2025-03-05:
I need the actual capture log, not a JPG.

lula6894, 2025-03-05:
Apologizes, I didn't explain well. Both attached files are capture files. As the attachment clip does not let me to attach .cap files, I renamed to .jpg to do it.
Please download both .jpg files and rename to .cap, you found the data there.
Sorry for the inconveniences. 
All the best

naugehyde, 2025-03-05:
Thanks.

The Voltage value appears to be embedded in the device name. Which would
put the voltage of your battery at 12.08 volts. Does that seem right?

lula6894, 2025-03-05:
The voltage is located in the device name. There is not an app or something to manage Lancol, just checking the voltage in the Bluetooth scan. 
Current voltage of my starter battery is 12.08 volts. Yes, it's a little bit low ... I need to charge.

naugehyde, 2025-03-05:
The Lancol class is ready for testing. 

To install execute the following from a command line:

`curl https://github.com/naugehyde/bt-sensors-plugin-sk/blob/1.1.0/sensor_classes/LancolVoltageMeter.js >  ~/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/LancolVoltageMeter.js`

To test, restart Signalk. Look for your Lancol device in the plugin configuration.

Let me know how it goes!

If all goes well, I'll package the class in the next revision.

lula6894, 2025-03-05:
I did it but bt-sensors-plugin disappears from my installed plugin list. I tried to reinstall but unsuccessfully.  
The log is here:
Mar 05 15:44:57 bt-sensors-plugin-sk failed to start: LancolVoltageMeter is not defined
Mar 05 15:44:57 ReferenceError: LancolVoltageMeter is not defined at Object.<anonymous> (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/LancolVoltageMeter.js:34:16) at Module._compile (node:internal/modules/cjs/loader:1734:14) at Object..js (node:internal/modules/cjs/loader:1899:10) at Module.load (node:internal/modules/cjs/loader:1469:32) at Function._load (node:internal/modules/cjs/loader:1286:12) at TracingChannel.traceSync (node:diagnostics_channel:322:14) at wrapModuleLoad (node:internal/modules/cjs/loader:235:24) at Module.require (node:internal/modules/cjs/loader:1491:12) at require (node:internal/modules/helpers:135:16) at /home/lula/.signalk/node_modules/bt-sensors-plugin-sk/index.js:81:17 at Array.forEach (<anonymous>) at Object.loadClasses (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/index.js:79:15) at loadClassMap (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/index.js:165:34) at module.exports (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/index.js:350:2) at doRegisterPlugin (/usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js:412:22) at registerPlugin (/usr/lib/node_modules/signalk-server/lib/interfaces/plugins.js:258:13)
Mar 05 15:44:57 signalk-server running at 0.0.0.0:[object Object]
Hope it helps

lula6894, 2025-03-05:
I check the code and I think there is a mistake in  lancolVoltageReader.
I changed to lancolVoltageMeter and now works.

lula6894, 2025-03-05:
I can see in the data browser the signal.strengh and the voltage, but the voltage is in "ratio" mode and measure 0.32
have a look to the screenshot

[image]

lula6894, 2025-03-05:
May be the log helps:
Mar 05 16:44:48 ReferenceError: the is not defined at LancolVoltageMeter.propertiesChanged (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/LancolVoltageMeter.js:30:13) at BusHelper.<anonymous> (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:384:26) at BusHelper.emit (node:events:507:28) at ProxyInterface.<anonymous> (/home/lula/.signalk/node_modules/node-ble/src/BusHelper.js:41:16) at ProxyInterface.emit (node:events:507:28) at ProxyListener.fn (/home/lula/.signalk/node_modules/dbus-next/lib/client/proxy-interface.js:19:18) at EventEmitter.emit (node:events:507:28) at handleMessage (/home/lula/.signalk/node_modules/dbus-next/lib/bus.js:122:23) at EventEmitter.<anonymous> (/home/lula/.signalk/node_modules/dbus-next/lib/bus.js:151:9) at EventEmitter.emit (node:events:507:28) at /home/lula/.signalk/node_modules/dbus-next/lib/connection.js:132:14 at Socket.<anonymous> (/home/lula/.signalk/node_modules/dbus-next/lib/message.js:65:9) at Socket.emit (node:events:507:28) at emitReadable_ (node:internal/streams/readable:832:12) at process.processTicksAndRejections (node:internal/process/task_queues:89:21)
Mar 05 16:45:02 ::1 - - [05/Mar/2025:16:45:02 +0000] "GET /skServer/logfiles/ HTTP/1.1" 304 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"

lula6894, 2025-03-05:
> I can see in the data browser the signal.strengh and the voltage, but the voltage is in "ratio" mode and measure 0.32 have a look to the screenshot
> 
> [image]

This was my mistake, voltage does not appears in this trial. I

lula6894, 2025-03-05:
There are still something to fix. Voltage does not appears on Data browser.
SignalK log:
Mar 05 17:15:30 TypeError: props.Name.match is not a function at LancolVoltageMeter.propertiesChanged (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/LancolVoltageMeter.js:30:59) at BusHelper.<anonymous> (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:384:26) at BusHelper.emit (node:events:507:28) at ProxyInterface.<anonymous> (/home/lula/.signalk/node_modules/node-ble/src/BusHelper.js:41:16) at ProxyInterface.emit (node:events:507:28) at ProxyListener.fn (/home/lula/.signalk/node_modules/dbus-next/lib/client/proxy-interface.js:19:18) at EventEmitter.emit (node:events:507:28) at handleMessage (/home/lula/.signalk/node_modules/dbus-next/lib/bus.js:122:23) at EventEmitter.<anonymous> (/home/lula/.signalk/node_modules/dbus-next/lib/bus.js:151:9) at EventEmitter.emit (node:events:507:28) at /home/lula/.signalk/node_modules/dbus-next/lib/connection.js:132:14 at Socket.<anonymous> (/home/lula/.signalk/node_modules/dbus-next/lib/message.js:65:9) at Socket.emit (node:events:507:28) at emitReadable_ (node:internal/streams/readable:832:12) at process.processTicksAndRejections (node:internal/process/task_queues:89:21)

naugehyde, 2025-03-05:
> EventEmitter. (/home/lula/.signalk/node_modules/dbus-next/lib/bus.js:151:9)
> Socket. (/home/lula/.signalk/node_modules/dbus-next/lib/message.js:65:9) at

One more time. Hopefully, the last!

lula6894, 2025-03-05:
Thanks to you for spending your time with me sensor and thanks for your programming knowledge.
I'll test just know

lula6894, 2025-03-05:
Mar 05 17:43:00 signalk-server running at 0.0.0.0:[object Object]
Mar 05 17:43:01 TypeError: this.getMetadatum(...).read is not a function at LancolVoltageMeter.emitData (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:558:47) at LancolVoltageMeter.propertiesChanged (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/LancolVoltageMeter.js:31:18) at LancolVoltageMeter.listen (/home/lula/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:579:18) at /home/lula/.signalk/node_modules/bt-sensors-plugin-sk/index.js:370:57 at process.processTicksAndRejections (node:internal/process/task_queues:105:5)

naugehyde, 2025-03-05:
once more.

This is actually a very simple sensor but it's the first one that's
required parsing of the Name property.

lula6894, 2025-03-05:
> One more time!
> 
> Thanks for going through this with me. I'm flying blind here.
> […](#)

I can give remote access to my system

lula6894, 2025-03-05:
Congrats!!! we have the voltage in Data Browser, unfortunately is null

[image]

No errors in the log

naugehyde, 2025-03-05:
Try the latest. Turn on debug. Send log if the data isn't coming up.

lula6894, 2025-03-05:
Still voltage in data browser is null.
But may be it's not important but the device in BT browser is "Lancol 12.08v" but in the plugin browser the name is "lancol-1" that is the name I setup in the plugit without voltage. Look at the picture.

[image]

The log:

Mar 05 18:26:11 ::1 - - [05/Mar/2025:18:26:11 +0000] "GET /admin/fonts/fa-solid-900.1551f4f60c37af51121f106501f69b80.woff2 HTTP/1.1" 304 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:26:11 ::1 - - [05/Mar/2025:18:26:11 +0000] "GET /signalk HTTP/1.1" 304 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:26:11 ::1 - - [05/Mar/2025:18:26:11 +0000] "GET /skServer/addons HTTP/1.1" 304 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:26:11 ::1 - - [05/Mar/2025:18:26:11 +0000] "GET /skServer/security/access/requests HTTP/1.1" 304 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:26:11 ::1 - - [05/Mar/2025:18:26:11 +0000] "GET /admin/fonts/signal-k-logo-image-text.d132ec2c4f0269c103c868aaa6868f01.svg HTTP/1.1" 304 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:26:11 ::1 - - [05/Mar/2025:18:26:11 +0000] "GET /admin/img/favicon.ico HTTP/1.1" 304 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:26:16 ::1 - - [05/Mar/2025:18:26:16 +0000] "POST /skServer/plugins/bt-sensors-plugin-sk/config HTTP/1.1" 200 53 "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:26:17 ::1 - - [05/Mar/2025:18:26:17 +0000] "POST /skServer/plugins/bt-sensors-plugin-sk/config HTTP/1.1" 200 53 "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:26:20 ::1 - - [05/Mar/2025:18:26:20 +0000] "GET /skServer/plugins HTTP/1.1" 200 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:26:21 2025-03-05T18:26:21.753Z bt-sensors-plugin-sk Connecting to bluetooth adapter hci0
Mar 05 18:26:21 2025-03-05T18:26:21.759Z bt-sensors-plugin-sk Starting scan...
Mar 05 18:26:21 2025-03-05T18:26:21.765Z bt-sensors-plugin-sk Plugin 1.1.0-beta.2.2.1.5 restarting...
Mar 05 18:26:21 2025-03-05T18:26:21.767Z bt-sensors-plugin-sk Waiting on A4:C1:38:2D:ED:07
Mar 05 18:26:21 2025-03-05T18:26:21.767Z bt-sensors-plugin-sk Waiting on A4:C1:38:DF:2D:37
Mar 05 18:26:21 2025-03-05T18:26:21.768Z bt-sensors-plugin-sk Waiting on A4:C1:38:E1:20:83
Mar 05 18:26:21 2025-03-05T18:26:21.768Z bt-sensors-plugin-sk Waiting on C8:FD:19:02:33:96
Mar 05 18:26:21 2025-03-05T18:26:21.772Z bt-sensors-plugin-sk Waiting on 13:05:AA:00:0B:B1
Mar 05 18:26:22 2025-03-05T18:26:22.770Z bt-sensors-plugin-sk Found A4:C1:38:2D:ED:07
Mar 05 18:26:22 2025-03-05T18:26:22.771Z bt-sensors-plugin-sk Found A4:C1:38:DF:2D:37
Mar 05 18:26:22 2025-03-05T18:26:22.775Z bt-sensors-plugin-sk Found A4:C1:38:E1:20:83
Mar 05 18:26:22 2025-03-05T18:26:22.776Z bt-sensors-plugin-sk Found C8:FD:19:02:33:96
Mar 05 18:26:22 2025-03-05T18:26:22.778Z bt-sensors-plugin-sk Found 13:05:AA:00:0B:B1
Mar 05 18:26:22 2025-03-05T18:26:22.780Z bt-sensors-plugin-sk instantiated A4:C1:38:2D:ED:07
Mar 05 18:26:22 2025-03-05T18:26:22.785Z bt-sensors-plugin-sk Listening for changes from lulaTAH-07 (A4:C1:38:2D:ED:07 RSSI: -71 db / 51%) ▂ ▄
Mar 05 18:26:22 2025-03-05T18:26:22.787Z bt-sensors-plugin-sk instantiated A4:C1:38:DF:2D:37
Mar 05 18:26:22 2025-03-05T18:26:22.787Z bt-sensors-plugin-sk instantiated A4:C1:38:E1:20:83
Mar 05 18:26:22 2025-03-05T18:26:22.794Z bt-sensors-plugin-sk Listening for changes from lulaTAH-37 (A4:C1:38:DF:2D:37 RSSI: -57 db / 72%) ▂ ▄ ▆
Mar 05 18:26:22 2025-03-05T18:26:22.794Z bt-sensors-plugin-sk Listening for changes from lulaTAH-83 (A4:C1:38:E1:20:83 RSSI: NaN db / NaN%)
Mar 05 18:26:22 2025-03-05T18:26:22.807Z bt-sensors-plugin-sk instantiated C8:FD:19:02:33:96
Mar 05 18:26:22 2025-03-05T18:26:22.809Z bt-sensors-plugin-sk v=null vf=NaN
Mar 05 18:26:22 2025-03-05T18:26:22.810Z bt-sensors-plugin-sk Listening for changes from lancol-1 (C8:FD:19:02:33:96 RSSI: -85 db / 23%) ▂
Mar 05 18:26:22 2025-03-05T18:26:22.930Z bt-sensors-plugin-sk Connecting to bluetooth adapter hci0
Mar 05 18:26:22 2025-03-05T18:26:22.937Z bt-sensors-plugin-sk Starting scan...
Mar 05 18:26:22 2025-03-05T18:26:22.952Z bt-sensors-plugin-sk Plugin 1.1.0-beta.2.2.1.5 restarting...
Mar 05 18:26:22 2025-03-05T18:26:22.954Z bt-sensors-plugin-sk Waiting on A4:C1:38:2D:ED:07
Mar 05 18:26:22 2025-03-05T18:26:22.954Z bt-sensors-plugin-sk Waiting on A4:C1:38:DF:2D:37
Mar 05 18:26:22 2025-03-05T18:26:22.955Z bt-sensors-plugin-sk Waiting on A4:C1:38:E1:20:83
Mar 05 18:26:22 2025-03-05T18:26:22.955Z bt-sensors-plugin-sk Waiting on C8:FD:19:02:33:96
[... cut ...]
Mar 05 18:30:55 2025-03-05T18:30:55.110Z bt-sensors-plugin-sk v=null vf=NaN
Mar 05 18:31:04 ::1 - - [05/Mar/2025:18:31:04 +0000] "GET /skServer/debugKeys HTTP/1.1" 304 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"
Mar 05 18:31:04 ::1 - - [05/Mar/2025:18:31:04 +0000] "GET /skServer/logfiles/ HTTP/1.1" 304 - "http://localhost:3000/admin/" "Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36"

naugehyde, 2025-03-05:
Please try again with latest code.

lula6894, 2025-03-05:
[image]

naugehyde, 2025-03-05:
> clap / clap / clap ... excellent

Let me know if it updates properly and if there are any other issues.

I made this extra hard on myself by multi-tasking some other work. Lesson
learned.

lula6894, 2025-03-05:
Thank you so much, it's a great improvement for my boat

lula6894, 2025-03-05:
Yes. Cadiz - Spain

naugehyde, 2025-03-05:
Awesome. Developing the plugin has allowed me to travel (virtually and
vicariously) all over the world!

lula6894, 2025-03-05:
Thanks again, keep in touch. I'll appreciate if you include "lancol" in the next version. I'll keep an eye on my battery voltage and report any issue. All the best my friend

lula6894, 2025-03-06:
Apologizes boring you with the same topic. I'm installing the plugin onboard
I installed through SignalK Appstore and add the LancolVoltageMeter.js file.
When I config the plugin I cannot select on the "sensor class" list and only appears "strength" and not "voltage". Have a look to the file. May be I did something wrong. 

[image]

naugehyde, 2025-03-06:
Looks like you just need to restart signal k.

lula6894, 2025-03-06:
Yes, I should reboot to make it works, restart SignalK was not enough. Thanks again and apologizes.

lula6894, 2025-04-04:
Hi again, I'm really happy with the Lancol addition to the plugin. Should be possible to add a "fine tunning" of the voltage?. 
Lancol is not a high accuracy voltmeter. When I make the battery measurement with a digital multimeter, there are a few milivolts of offset. I was wondering if should be possible to compensate it in the plugin setup. It's not important, urgent or necessary, just an improvement in any of the updates.
Thanks again for your super plugin, all the best

naugehyde, 2025-04-04:
Hi Lula,

Try the Calibration plugin. I think it will do what you're looking for.

lula6894, 2025-04-05:
Apologizes for the delay answering. Definitely right, "calibration" is what I need. Thanks again