naugehyde/bt-sensors-plugin-sk #27: Non continous reading from Inkbird thermometer

CasparSprungli, 2025-03-24:
Hi, 
Thanks for developing this plugin.

I am setting up an Inkbird IBS-TH2 on my Raspberry Pi and only get realtime DB data.    To get TEMP readings, I need to restart the SignalK server.

Is there a workarround?  Am I missing something?

CasparSprungli, 2025-03-24:
Hi,

thanks for your prompt reply.   

What I mean by realtime DB data is that the RSSI information gets updated on SignalK every other second.

1) This is the bluetoothctl info

pi@openplotter:~ $ bluetoothctl info 49:24:03:04:02:99
Device 49:24:03:04:02:99 (public)
        Name: tps
        Alias: tps
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (0000fff0-0000-1000-8000-00805f9b34fb)
        ManufacturerData Key: 0x0906
        ManufacturerData Value:
  00 00 00 af 9c 64 08                             .....d.
        ManufacturerData Key: 0x08fc
        ManufacturerData Value:
  00 00 00 76 74 64 08                             ...vtd.
        ManufacturerData Key: 0x08f2
        ManufacturerData Value:
  00 00 00 1f b5 64 08                             .....d.
        ManufacturerData Key: 0x08e8
        ManufacturerData Value:
  00 00 00 46 77 64 08                             ...Fwd.
        ManufacturerData Key: 0x08c0
        ManufacturerData Value:
  00 00 00 26 71 64 08                             ...&qd.
        ManufacturerData Key: 0x08a2
        ManufacturerData Value:
  00 00 00 df b9 64 08                             .....d.
        ManufacturerData Key: 0x088e
        ManufacturerData Value:
  00 00 00 4e 7f 64 08                             ...N.d.
        ManufacturerData Key: 0x0884
        ManufacturerData Value:
  00 00 00 d6 7e 64 08                             ....~d.
        ManufacturerData Key: 0x087a
        ManufacturerData Value:
  00 00 00 ff aa 64 08                             .....d.
        ManufacturerData Key: 0x092e
        ManufacturerData Value:
  00 00 00 cf 9a 64 08                             .....d.
        ManufacturerData Key: 0x0938
        ManufacturerData Value:
  00 00 00 86 59 64 08                             ....Yd.
        ManufacturerData Key: 0x094c
        ManufacturerData Value:
  00 00 00 36 52 64 08                             ...6Rd.
        ManufacturerData Key: 0x0956
        ManufacturerData Value:
  00 00 00 6f 90 64 08                             ...o.d.
        ManufacturerData Key: 0x0960
        ManufacturerData Value:
  00 00 00 a7 94 64 08                             .....d.
        RSSI: -30
pi@openplotter:~ $

2) In the last 3 hours, I got TEMP readings at 11:15, 11:26, 11:46, 12:19, 12:37, 12:47 and nothing after that.  Now been 14:17
[image]

3) Inkbird is sitting side by side with the RPI

4) Batteries are new.  On the Android App, it reads 100% battery

5)   This is the DebugLog for the last hour:

[Mar 24 141510 GET skServerplugins 2.txt](https://github.com/user-attachments/files/19434850/Mar.24.141510.GET.skServerplugins.2.txt)

CasparSprungli, 2025-03-24:
pi@openplotter:~ $ bluetoothctl show
Controller 2C:CF:67:88:AA:3B (public)
        Name: openplotter
        Alias: openplotter
        Class: 0x006c0000
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0542
        Discovering: yes
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
pi@openplotter:~ $

naugehyde, 2025-03-24:
Definitely not the scanner.

The inkbird works like many bt temp sensors by broadcasting its data. It does so sporadically and comprehensively broadcasting its entire temp history. My plugin can not update its path without a broadcast. 

FWIW, the device appears not to broadcast its temp if the current temp has changed -- likely to save battery power. It's RSSI broadcast is considerably more frequent.

It is a completely connection-less protocol. The Inkbird app may have a connection protocol that assures more timely broadcasts but it is not documented. 

You may be seeing the temp settle at a value and not rebroadcast to save battery.

A simple test is to hold the TH2 or place a cold object (like a can of soda) atop it. 

You can watch the data stream by executing

`
bluetoothctl | grep <mac> 
`

If you don't see a stream of "Manufacturer Data" coming from the TH2 that means it's not broadcasting it's temp -- probably because it hasn't changed since the last broadcast.

naugehyde, 2025-03-24:
So there is a GATT connection protocol for the Inkbird that will give realtime data like their app does.  (see https://github.com/SteidlD/homebridge-inkbird-bt-thsensor/blob/master/index.js)

If you want something like that implemented for SK, create an enhancement issue or better yet, if you are adept at nodejs, create a pull request. 

NOTE: GATT connections tends to drain batteries. FWIW, the plugin is designed to connect and read GATT data at prescribed intervals (if implemented) which will get you realtime data -- just not continuously. 

IMO, a realtime continuous stream is probably best handled by the Inkbird app. But if I have time, I'll look into improving support for the inkbird-- it's a question of time and priority.

CasparSprungli, 2025-03-24:
> Note: The Inkbird is delayed in its temp change broadcasts by about a minute. It was designed for more or less constant temp environments like refrigerators and freezers.
> 
> If you're looking for true real-time data, there are better options (but they are probably not battery powered)

I am not looking for real-time data, since I will place the Inkbird in the fridge.  However, it intrigued me the delay in the temperature data.  You are right, it broadcasts temp at odd intervals.

Thanks again for answering!

naugehyde, 2025-03-24:
Great! 

I'll close this but I'll put a note up on the config page for the Inkbird about it's potentially long reporting delays. 

Thanks for letting me know about the issue.

closed 2025-03-24 by naugehyde