naugehyde/bt-sensors-plugin-sk #28: Beta 2.2.3 won't install on Raspberry Pi

JohnStov, 2025-03-27:
Installation log below:

[2025-03-27T17_32_56_979Z-debug-0.log](https://github.com/user-attachments/files/19495217/2025-03-27T17_32_56_979Z-debug-0.log)

naugehyde, 2025-03-27:
The error originates from usocket which as far as I know my plugin doesn't use.

`1882 verbose reify failed optional dependency /home/pi/.signalk/node_modules/usocket
`
Not sure why you're getting that error or the subsequent ones. FWIW the invocation of npm is also loading a variety of plugins (Kip etc.) I suspect you'd get these errors no matter what module you attempted to install.

FWIW, no other users are reporting this error.

I'm no expert in npm but it appears something is wrong with your installation. Regrettably, I can't provide any guidance to help you fix it.

In the past when I have gotten stuck with weird errors like this, I've ended up reinstalling SignalK.   

If you discover the root issue and a good solution that doesn't involve reinstalling SK, please document it here.

Thank you and fair winds!

naugehyde, 2025-03-27:
This may be helpful:

https://docs.npmjs.com/cli/v9/commands/npm-rebuild

JohnStov, 2025-03-31:
Reinstalled SignalK and everything is fine now

closed 2025-03-31 by JohnStov