naugehyde/bt-sensors-plugin-sk #29: beta.2.2.3 drops reporting frequency

Peter-Petrik, 2025-03-29:
beta.2.2.2 reports values about every 2 seconds. Upgrading to beta.2.2.3 drops the frequency to every 40-50 seconds. I don't see any new settings reflecting this, so I assume it's a bug rather than a feature?

Reinstalling beta.2.2.2 resolves the issue.

naugehyde, 2025-03-29:
Hi Peter.

Which device and which paths are reporting at longer intervals?

Peter-Petrik, 2025-03-29:
Victron SmartSolar MPPT is where I noticed it, for path electrical.solar.port.panelPower

naugehyde, 2025-03-29:
There was a change that I made that might effect the reporting frequency-- it's apparently a side effect of Bluez's Transport scan option as it happens. 

I'll commit a change that you can clone or manually patch (it's just one file) so you can test it. Are you okay with that?

Peter-Petrik, 2025-03-29:
Sounds great. Happy to pull a specific commit and test.

naugehyde, 2025-03-29:
Upgrade to latest then execute from a command line:

`curl https://github.com/naugehyde/bt-sensors-plugin-sk/raw/refs/heads/1.1.0/index.js  > ~/.signalk/node_modules/bt_sensors_plugin_sk`

If all goes well you should see a Transport dropdown in the plugin config. Experiment with the various options and let me know the results. 

Thanks.

naugehyde, 2025-03-29:
Give me about an hour. Bug in latest commit.

naugehyde, 2025-03-29:
Try it now.

The only two relevant options for Transport are LE and Auto. BR/EDR will probably fail to find new devices.

Peter-Petrik, 2025-03-29:
Tested the latest commit https://github.com/naugehyde/bt-sensors-plugin-sk/commit/612c7c5124114fe6ede982eddf544683763b4c12, and with Transport set to "LE" the update frequency is back to ~2s intervals.

Appreciate your prompt responses!

naugehyde, 2025-03-29:
Can you test Auto as well?

Peter-Petrik, 2025-03-29:
`Auto` seems to update every 7-8s.

naugehyde, 2025-03-29:
Okay, thanks!

closed 2025-04-13 by naugehyde