naugehyde/bt-sensors-plugin-sk #30: Beta 2.2.3 don't read my Xiaomi LYWSD03MMC

lula6894, 2025-03-30:
Beta 2.2.2 works fine with Xiaomi LYWSD03MMC Temperature & Humidity devices flashed with ATC firmware.
Now 2.2.3 read them just when signalK starts, but it's not pulling them.

naugehyde, 2025-03-30:
What happens if you increase the Discovery Timeout for the devices?

naugehyde, 2025-03-30:
> Updated to 60 sec.
> This is a part of the log:
> Mar 30 12:00:04 2025-03-30T12:00:04.799Z bt-sensors-plugin-sk Sensor at
> A4:C1:38:DF:2D:37 unavailable. Reason: operation timed out
> Mar 30 12:00:04 2025-03-30T12:00:04.799Z bt-sensors-plugin-sk operation
> timed out
> Mar 30 12:00:04 2025-03-30T12:00:04.801Z bt-sensors-plugin-sk Unable to
> communicate with device A4:C1:38:E1:20:83 Reason: operation timed out
> Mar 30 12:00:04 2025-03-30T12:00:04.801Z bt-sensors-plugin-sk Error:
> operation timed out at Timeout.
> (/home/lula/.signalk/node_modules/node-ble/src/Adapter.js:158:16) at
> listOnTimeout (node:internal/timers:611:17) at process.processTimers
> (node:internal/timers:546:7)
> Mar 30 12:00:04 2025-03-30T12:00:04.801Z bt-sensors-plugin-sk Sensor at
> A4:C1:38:E1:20:83 unavailable. Reason: operation timed out
> Mar 30 12:00:04 2025-03-30T12:00:04.801Z bt-sensors-plugin-sk operation
> timed out

Try the latest package in NPM (2.2.4)

lula6894, 2025-03-30:
2.2.4 - erratic behavior, the BT devices disappear from the plugin setup, or appears without signalk data setup .
I'm using three Xiaomi T&H devices, all with the same behavior (In 2.2.2 working ok). In 2.2.3 and 2.2.4 Lancol works

naugehyde, 2025-03-30:
The erratic behavior is nothing new. The config screen, for the time being,
needs you to be patient while it finds, or times out finding, all the
devices you've configured. I'm working on a solution right now.

In the meantime, turn debug on. Keep your eye on the server log. When all
of your devices are found by the server's Bluetooth device, refresh the
config page.

lula6894, 2025-03-30:
Ok, I'll do it and keep you informed

naugehyde, 2025-03-30:
Thank you.

A custom configuration screen that will make life easier for everyone is on
the way. Note: I'm in early stages of development on it.

lula6894, 2025-03-30:
Thanks, it's a great plugin, in particular nowadays with a lot of low-cost BT devices available.

naugehyde, 2025-03-30:
Thanks. Availability of inexpensive BT devices was a huge part of my
motivation for writing the plugin. The first device I supported WAY back
when before I published the plugin was a Xiaomi thermometer. The choice
between a $5 wireless thermometer and a $150 *wired* thermometer for my
refrigerator was an easy one!

lula6894, 2025-03-30:
I own three, one for the freezer, one for the fridge and one for the ambient to drive a dehumidifier.
Now I'm working in a new LiFePo4 installation and almost all BMS has BT. Your plugin is a must to access from everywhere

naugehyde, 2025-03-30:
FYI, I was reluctant to release the plugin with the config screen as it is.
It's gratifying to know that it's still proving useful.

closed 2025-04-13 by naugehyde