naugehyde/bt-sensors-plugin-sk #32: Running in docker

macjl, 2025-04-13:
Hello,

Do you know what are the steps to run this plugin in docker ? I run it in privileged mode, and have some errors : 

```
POST /skServer/plugins/bt-sensors-plugin-sk/config 200 7.786 ms - 53
DBusError: Failed to execute program org.bluez: Permission denied
    at _methodReturnHandlers.<computed> (/home/node/.signalk/node_modules/dbus-next/lib/bus.js:343:27)
    at handleMessage (/home/node/.signalk/node_modules/dbus-next/lib/bus.js:101:11)
    at EventEmitter.<anonymous> (/home/node/.signalk/node_modules/dbus-next/lib/bus.js:151:9)
    at EventEmitter.emit (node:events:524:28)
    at /home/node/.signalk/node_modules/dbus-next/lib/connection.js:132:14
    at Socket.<anonymous> (/home/node/.signalk/node_modules/dbus-next/lib/message.js:65:9)
    at Socket.emit (node:events:524:28)
    at emitReadable_ (node:internal/streams/readable:834:12)
    at process.processTicksAndRejections (node:internal/process/task_queues:81:21) {
  type: 'org.freedesktop.DBus.Error.Spawn.ExecFailed',
  text: 'Failed to execute program org.bluez: Permission denied',
  reply: Message {
    type: 3,
    _sent: false,
    _serial: 4,
    path: undefined,
    interface: undefined,
    member: undefined,
    errorName: 'org.freedesktop.DBus.Error.Spawn.ExecFailed',
    replySerial: 3,
    destination: ':1.1',
    sender: 'org.freedesktop.DBus',
    signature: 's',
    body: [ 'Failed to execute program org.bluez: Permission denied' ],
    flags: 1
  }
} 
```

macjl, 2025-04-13:
I just needed to add /var/run/dbus as mount point, and it worked. Here is my (simplified) docker-compose.yml : 

```

services:
  signalk:
    image: signalk/signalk-server
    volumes:
      - ~/signalk:/home/node/.signalk
      - /var/run/dbus:/var/run/dbus
    ports:
      - 3000:3000
      - 10110:10110
    privileged: true
```

naugehyde, 2025-04-13:
Excellent. Thank you!

closed 2025-04-13 by naugehyde