naugehyde/bt-sensors-plugin-sk #34: ATC temp/humidity stopped being picked up

isarychkin, 2025-04-25:
Hi there,
test platforms: 
1) rpi 3b running bookworm with signalk 2.13.5
2) rpi 3b - victron venus os 3.60 signalk ~ 2.13

alpha versions of the plug-in used to work fine with the ATC temp/humidity sensors ( A4:C1:38:*:*:*) 
The current release of 1.1.0 picks up myriads of ble devices except the ATC ble sensor even when it sits right next to RPi.
I know the ATC sensors are working as I can still pick them up via an ESP32 bridge.

Please bring back the alpha versions ;)

Cheers

naugehyde, 2025-04-25:
I have an ATC thermometer on my boat and it works fine with all versions of the plugin.

If the plugin is not picking up your ATC there are three possibilities that come to mind. 

1) Timeout value. Even though it's right there next to your server, the ATC is optimized for battery life and advertises itself infrequently. Open the plugin config and increase the timeout value for the device to 60s or more until it reliably finds it on startup.

2) The device is connected via GATT to another app. I've noticed that with the stock Xiaomi firmware the device doesn't advertise itself at all if it is connected via GATT to the MiHome app. Not sure about ATC and GATT connections but it's a possible explanation.

3) The scanner on your server has been turned off by another application and/or the scanner is in an inconsistent state and not working. I've seen this occur and it's addressed in the next release. To test if the scanner is on or not, go to the command line and execute ``bluetoothctl``. If you see data flying by, the scanner is on and working. 

Please try changing the Timeout value first.

If that doesn't work please send your config file and a debug log (after turning debug on for the plugin).

To create a debug log execute the following from a command line:

``journalctl -u signalk.service --since Today > ~/isarychkin.debug.log ``

isarychkin, 2025-04-27:
Thank you so much for your help!  Got one rPi SignalK (on plain bookworm img) working with my modded ATC sensor by increasing scan timeout to 300.  Can remove esp32 from loop now.  Super!  
Away from my boat atm so can't test the one with Venus OS Large image.
Cheers.

naugehyde, 2025-04-27:
Great to hear!

Always happy to help.

FYI I'm currently experimenting with USB Bluetooth dongles that have stronger Receive signals.

Best,

Andy

closed 2025-04-27 by naugehyde