naugehyde/bt-sensors-plugin-sk #35: Govee H5175 Temp Sensor Detected as iBeacon

zambaras, 2025-04-27:
After updating to the latest version of the BT-Sensors plugin, my Govee H5103 (GV5179_3F73) is being incorrectly classified as an iBeacon rather than a temperature/humidity sensor.

Previously, it was correctly parsed as a Govee temperature/humidity device and reported environment.temperature and environment.humidity in SignalK.

Now, when it's detected and I add it, the plugin only provides fields to map for RSSI and battery.

I confirmed via the Govee Home app that the sensor is still correctly broadcasting temperature and humidity.

I'm going to try rolling back the plugin.

Capt Frank

naugehyde, 2025-04-27:
I noticed this too. (The iBeacon was added after a pull request. Going to
move it to a separate package as it's problematic with BTSensor::identify)

Which version are you working with?

naugehyde, 2025-04-27:
Don't worry about the version. Looks like the IBeacon identify method is
broken and potentially mis-identifies *any* sensor that supports its own
protocol as well as the IBeacon protocol.

Will get a fix out ASAP.

naugehyde, 2025-04-27:
Release 1.1.1 of the plugin with the misidentification fix in place.

Tested it at home with some Govee devices.

Please let me know if it solves your problem and I'll close the issue.

zambaras, 2025-04-27:
Amazing.  This did the trick.  Thanks for your work on this great project!

naugehyde, 2025-04-27:
Great to hear. Thanks for letting me know! Closing the issue now.

closed 2025-04-27 by naugehyde