naugehyde/bt-sensors-plugin-sk #37: XIAOMI temperature/humidity LYWSD03MMC Unable to get service data

guthip, 2025-05-06:
Succesfully using the BT plugin for 3x Victron devices (MPPT 75/15, AC charger IP43, DCDC charger Orion-Tr Smart 12|12-30).

Stumped after adding a XIAOMI temperature/humidity device (type LYWSD03MMC). With debug mode in the plugin active, this is the message:

```
May 05 17:45:01 2025-05-06T00:45:01.630Z bt-sensors-plugin-sk Error occured on xiaomi0 at A4:C1:38:2E:EA:9A: Unable to get service data for xiaomi0 (A4:C1:38:2E:EA:9A RSSI: -49 db / 81%) ▂ ▄ ▆ █ 
May 05 17:45:01 2025-05-06T00:45:01.631Z bt-sensors-plugin-sk Error: Unable to get service data for xiaomi0 (A4:C1:38:2E:EA:9A RSSI: -49 db / 81%) ▂ ▄ ▆ █ at ATC.propertiesChanged (/home/fgd/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/ATC.js:58:19) at BusHelper.<anonymous> (/home/fgd/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:384:26) at BusHelper.emit (node:events:524:28) at ProxyInterface.<anonymous> (/home/fgd/.signalk/node_modules/node-ble/src/BusHelper.js:41:16) at ProxyInterface.emit (node:events:524:28) at ProxyListener.fn (/home/fgd/.signalk/node_modules/dbus-next/lib/client/proxy-interface.js:19:18) at EventEmitter.emit (node:events:524:28) at handleMessage (/home/fgd/.signalk/node_modules/dbus-next/lib/bus.js:122:23) at EventEmitter.<anonymous> (/home/fgd/.signalk/node_modules/dbus-next/lib/bus.js:151:9) at EventEmitter.emit (node:events:524:28) at /home/fgd/.signalk/node_modules/dbus-next/lib/connection.js:132:14 at Socket.<anonymous> (/home/fgd/.signalk/node_modules/dbus-next/lib/message.js:65:9) at Socket.emit (node:events:524:28) at emitReadable_ (node:internal/streams/readable:834:12) at process.processTicksAndRejections (node:internal/process/task_queues:81:21)
```

guthip, 2025-05-06:
~/.signalk/plugin-config-data/bt-sensors-plugin-sk.json
attached

[bt-sensors-plugin-sk.json](https://github.com/user-attachments/files/20051656/bt-sensors-plugin-sk.json)

naugehyde, 2025-05-06:
Did you flash the ATC? Which custom firmware did you use?

guthip, 2025-05-06:
> Did you flash the ATC? Which custom firmware did you use?
> […](#)

The device has been flashed with the latest ATC_Exploit.bin from here:
https://github.com/atc1441/ATC_MiThermometer

guthip, 2025-05-06:
bluetoothctl info 38:3B:26:5A:6B:F4 > seb_bt.cap
bluetoothctl scan on >> seb_bt.cap

Results in

[XIAOMI_bt.txt](https://github.com/user-attachments/files/20051759/XIAOMI_bt.txt)

naugehyde, 2025-05-06:
Ah. Looks like the developers moved the service data to a different Service
ID. Yay.

Easy fix. Probably.

Which version of the plugin id are you using?

guthip, 2025-05-06:
> Ah. Looks like the developers moved the service data to a different Service ID. Yay. Easy fix. Probably. Which version of the plugin id are you using?
> […](#)

bt-sensors-plugin-sk version 1.1.1

guthip, 2025-05-06:
Temporary fix in
https://github.com/naugehyde/bt-sensors-plugin-sk/blob/main/sensor_classes/ATC.js , line 56
>> const buff = this.getServiceData("0000fcd2-0000-1000-8000-00805f9b34fb")
Does give results now, yeah!

naugehyde, 2025-05-06:
Oh cool! You did it yourself! I'll get a fix out that works for both
versions of ATC.

guthip, 2025-05-06:
.. but also an error in the log...  
```
May 05 20:13:11 
2025-05-06T03:13:11.611Z bt-sensors-plugin-sk Error occured on xiaomi0 
at A4:C1:38:2E:EA:9A: The value of "offset" is out of range. It must be 
>= 0 and <= 8. Received 10

May 05 20:13:11
 2025-05-06T03:13:11.611Z bt-sensors-plugin-sk RangeError 
[ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be 
>= 0 and <= 8. Received 10
    at boundsError (node:internal/buffer:88:9)
    at Buffer.readUInt16LE (node:internal/buffer:245:5)
    at Metadatum.read 
(/home/fgd/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/ATC.js:32:40)
    at ATC.emitData 
(/home/fgd/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:558:47)
    at ATC.propertiesChanged 
(/home/fgd/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/ATC.js:62:14)
    at BusHelper.<anonymous> 
(/home/fgd/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:384:26)
    at BusHelper.emit (node:events:524:28)
    at ProxyInterface.<anonymous> 
(/home/fgd/.signalk/node_modules/node-ble/src/BusHelper.js:41:16)
    at ProxyInterface.emit (node:events:524:28)
    at ProxyListener.fn 
(/home/fgd/.signalk/node_modules/dbus-next/lib/client/proxy-interface.js:19:18)
    at EventEmitter.emit (node:events:524:28)
    at handleMessage 
(/home/fgd/.signalk/node_modules/dbus-next/lib/bus.js:122:23)
    at EventEmitter.<anonymous> 
(/home/fgd/.signalk/node_modules/dbus-next/lib/bus.js:151:9)
    at EventEmitter.emit (node:events:524:28)
    at 
/home/fgd/.signalk/node_modules/dbus-next/lib/connection.js:132:14
    at Socket.<anonymous> 
(/home/fgd/.signalk/node_modules/dbus-next/lib/message.js:65:9)
```

guthip, 2025-05-06:
As expected, we only see the signal strength, humidity and temperature apparently correctly parsed

[Screen Shot 2025-05-05 at 8.15.26 PM.jpg.pdf](https://github.com/user-attachments/files/20052378/Screen.Shot.2025-05-05.at.8.15.26.PM.jpg.pdf)

naugehyde, 2025-05-06:
Do you know where the code is for the ATC_exploit.bin file?

guthip, 2025-05-06:
I don't know how he created ATC_Exploit.bin. 

But hang on. I originally started on the above repo to reprogram the sensor, then switched to this repo as was recommended by the first repo:
https://github.com/pvvx/ATC_MiThermometer/tree/master 

From that I uploaded firmware "bin/ATC_v51.bin" , if I remember well.
https://github.com/pvvx/ATC_MiThermometer/blob/master/bin/ATC_v51.bin

I'm looking at that now..

naugehyde, 2025-05-06:
Did you flash ATC_V51 or ATC_Exploit?

guthip, 2025-05-06:
Just confirmed I used ATC_V51

naugehyde, 2025-05-06:
> To make it more 'interesting', this is a parameter:
>
>    - Advertising type: "BTHome v2
>    -
>
>    other options are "ATC1441", "PVVX (custom)", and "MIJIA (MiHome)"
>
>
> Not sure what these do exactly, other than make the debug challenge larger.

I think I support  ATC1441 already, if you want to give that a try.

guthip, 2025-05-06:
Trying ATC1441 from the sensor, but still what seems to be the same out of bounds error message. I tried ATC_LE and ATC_BE modes in the plugin, and on/off the AdFlags in the sensor FW. No change in behavior detected.

[Screen Shot 2025-05-05 at 9.49.08 PM.jpg.pdf](https://github.com/user-attachments/files/20052908/Screen.Shot.2025-05-05.at.9.49.08.PM.jpg.pdf)

[Screen Shot 2025-05-05 at 9.50.26 PM.jpg.pdf](https://github.com/user-attachments/files/20052916/Screen.Shot.2025-05-05.at.9.50.26.PM.jpg.pdf)

guthip, 2025-05-06:
Note also that the values for humidity and temperature are not correctly parsed. It's warm in California, but not 700 F (400 C).

naugehyde, 2025-05-06:
If you can tell me how to parse

  40 00 65 01 50 02 d5 0a 03 80 14

I can implement it easily.

All I can see right now is 16bit LE at index 4 = humidity as ratio * 1000
(592)

And 16bit LE at index 6 = Temp in C * 100 (2773)

Can you confirm?

guthip, 2025-05-06:
Thanks for that hint. This works w/o a problem. Great time to get some sleep!

naugehyde, 2025-05-06:
Awesome!

(I'm in MDR and Venice. Where are you?)

guthip, 2025-05-06:
SF Bay area

guthip, 2025-05-06:
Thanks for all your help. The setup works fine now.

[Databrowser.pdf](https://github.com/user-attachments/files/20067337/Databrowser.pdf)

For the record, this is the configuration that works successfully:

- Device: LYWSD03MMC
- HW rev: B1.6
- SW rev: 0130
- FW rev: 1.0.0_0130
https://github.com/pvvx/ATC_MiThermometer/blob/master/bin/Original/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin 

With the flasher used from here: https://github.com/pvvx/ATC_MiThermometer/tree/master 

And a temporary fix for the serviceData

[Temp fix.pdf](https://github.com/user-attachments/files/20067346/Temp.fix.pdf)

naugehyde, 2025-05-06:
I'll add support for the 0000FCD2 service data packets.

Can you check to see if there's a reliable way to determine what FW level
the ATC is running on?

guthip, 2025-05-06:
> I'll add support for the 0000FCD2 service data packets. Can you check to see if there's a reliable way to determine what FW level the ATC is running on?
> […](#)

The aforementioned flasher shows the FW version installed after the device is connected.

naugehyde, 2025-05-06:
Sorry. I meant: is there a reliable way for the plugin to automatically
determine the FW version?

guthip, 2025-05-07:
@pvvx (victor), the developer of the FW may be able to shed to some light on that.

pvvx, 2025-05-07:
https://github.com/atc1441/ATC_MiThermometer - does not have any identifiers and standard UUIDs with HW and SW numbers.

All BLE/Zigbee FWs from https://github.com/pvvx/.... have standard UUID services with firmware and HW numbers.

Device Information Service
PrimaryService: 0x180a - Device Information
Characteristic: 0x2a24  - Model Number String
Characteristic: 0x2a25  - Serial Number String
Characteristic: 0x2a26  - Firmware Revision String
Characteristic: 0x2a27 -  Hardware Revision String
Characteristic: 0x2a28  - Software Revision String
Characteristic: 0x2a29  - Manufacturer Name String

It is possible to query version numbers and other information using a BLE connection.

There are 4 BLE beacon formats in use. And each has a second encrypted version. There are 8 formats in total.

In new versions of FW all old formats will be removed from summer of this year. Only official, registered and paid every year in Bluetooth SIG format will remain - BTHome ver 2.  (This warning is already 1.5 years old.)

closed 2025-06-29 by naugehyde