naugehyde/bt-sensors-plugin-sk #38: Victron SmartSolar charger

emmanuelcohenlaroque, 2025-05-07:
Greetings,
First of all, thanks a lot for this great work!

I have issues connecting a Victron SmarSolar device, here is the log. It seems that the encryption key is missing but this device does not offer such function using the Victron app. Have you had such experience before ?

`May 07 10:27:03 2025-05-07T08:27:03.410Z bt-sensors-plugin-sk Error: Unable to read data from Victron VictronSolarCharger (Model ID:41077) (CA:AB:A2:51:CB:6A RSSI: -71 db / 51%) ▂ ▄ : RangeError: Invalid key length at VictronSolarCharger.propertiesChanged (/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/Victron/VictronSensor.js:100:19) at BusHelper.<anonymous> (/home/pi/.signalk/node_modules/bt-sensors-plugin-sk/BTSensor.js:384:26) at BusHelper.emit (node:events:514:28) at ProxyInterface.<anonymous> (/home/pi/.signalk/node_modules/node-ble/src/BusHelper.js:41:16) at ProxyInterface.emit (node:events:514:28) at ProxyListener.fn (/home/pi/.signalk/node_modules/dbus-next/lib/client/proxy-interface.js:19:18) at EventEmitter.emit (node:events:514:28) at handleMessage (/home/pi/.signalk/node_modules/dbus-next/lib/bus.js:122:23) at EventEmitter.<anonymous> (/home/pi/.signalk/node_modules/dbus-next/lib/bus.js:151:9) at EventEmitter.emit (node:events:514:28) at /home/pi/.signalk/node_modules/dbus-next/lib/connection.js:132:14 at Socket.<anonymous> (/home/pi/.signalk/node_modules/dbus-next/lib/message.js:65:9) at Socket.emit (node:events:514:28) at emitReadable_ (node:internal/streams/readable:609:12) at process.processTicksAndRejections (node:internal/process/task_queues:81:21)`

naugehyde, 2025-05-07:
Hi Emmanuel.

The error occurs when the encryption key in the device configuration is not
properly set.

To find your device's encryption key follow the instructions from Victron
here:

https://www.victronenergy.com/media/pg/VictronConnect_Manual/en/stored-trends---instant-readout.html#UUID-a249901a-7ebe-9a66-3146-b9f5bb5c77af

naugehyde, 2025-05-08:
> Thanks a lot! Unfortunately, it seems that the encryption key cannot be
> obtained from Victron connect app for this device (MMPT75/15 SmartSolar
> charger) so I am a bit unfortunate! I have requested Victron if I can get
> access to those BLE properties in some other ways.

Hmm. Little surprised by that!

There's a Beta of the Victron Connect you can try:

https://www.victronenergy.com/live/victronconnect:beta

closed 2025-07-12 by naugehyde