naugehyde/bt-sensors-plugin-sk #39: incorrect version check

nmbath, 2025-05-14:
I believe that this lines

https://github.com/naugehyde/bt-sensors-plugin-sk/blob/c3a2561acbddbd52962dfe6a2d48666cf14636b5/index.js#L174

should actually be

semver.gte(app.config.version,"2.13.5")?"dist":"lib" 

The move from dist to lib  has already happened in 2.13.5 and am getting log error messages finding modules.js

naugehyde, 2025-05-14:
Yikes. Not sure how that got past me. 

Fix coming ASAP.

naugehyde, 2025-05-14:
Actually, hold on. That's working as designed. 

The module directory moved after 2.13.5. 

What error are you getting? Send trace please.

nmbath, 2025-05-14:
I have just downloaded a fresh signal-server clone and the release is 2.13.5 and the directory is dist and not lib. Therefore the  modules.js is not being found. They have not increased the version since March.

May 14 09:53:56 Error [ERR_MODULE_NOT_FOUND]: Cannot find module '/usr/lib/node_modules/signalk-server/lib/modules.js' imported from /usr/lib/node_modules/signalk-server/node_modules/bt-sensors-plugin-sk/index.js at finalizeResolution (node:internal/modules/esm/resolve:275:11) at moduleResolve (node:internal/modules/esm/resolve:860:10) at defaultResolve (node:internal/modules/esm/resolve:984:11) at ModuleLoader.defaultResolve (node:internal/modules/esm/loader:780:12) at #cachedDefaultResolve (node:internal/modules/esm/loader:704:25) at ModuleLoader.resolve (node:internal/modules/esm/loader:687:38) at ModuleLoader.getModuleJobForImport (node:internal/modules/esm/loader:305:38) at onImport.tracePromise.__proto__ (node:internal/modules/esm/loader:643:36) at TracingChannel.tracePromise (node:diagnostics_channel:344:14) at ModuleLoader.import (node:internal/modules/esm/loader:642:21) at defaultImportModuleDynamicallyForScript (node:internal/modules/esm/utils:234:31) at importModuleDynamicallyCallback (node:internal/modules/esm/utils:256:12) at loadClassMap (/usr/lib/node_modules/signalk-server/node_modules/bt-sensors-plugin-sk/index.js:177:17) at module.exports (/usr/lib/node_modules/signalk-server/node_modules/bt-sensors-plugin-sk/index.js:389:2) at doRegisterPlugin (/usr/lib/node_modules/signalk-server/dist/interfaces/plugins.js:397:22) at process.processTicksAndRejections (node:internal/process/task_queues:105:5) { code: 'ERR_MODULE_NOT_FOUND', url: 'file:///usr/lib/node_modules/signalk-server/lib/modules.js' }

naugehyde, 2025-05-14:
The clone of the source tree is not yet versioned (it will be 2.13.6 or
greater, currently it's at 2.13.5). The release version on npm (also
2.13.5) keeps its modules in lib not dist. Can't break the release for the
development version!

You'll have to manually change my plugin code or the main
branch's package.json until the main branch version is normalized.

nmbath, 2025-05-14:
ahh OK. Will apply local patch for the time being. Sorry. for the confusion.

closed 2025-05-14 by naugehyde