naugehyde/bt-sensors-plugin-sk #42: Request for device support: Litime

guthip, 2025-05-30:
I'm interested in added functionality for the Litime battery

LiTime 12V 100Ah Group 24 Bluetooth LiFePO4 Lithium Deep Cycle Battery
SKU:L12V100-100-GROUP24-BT-4-A70

There is reference code out there:
- https://github.com/calledit/LiTime_BMS_bluetooth/blob/main/index.html#L257
- https://github.com/chadj/litime-bluetooth-battery/blob/main/index.html

Anyone else interested? Anyone working on a plugin?
That's 2x yes for me, but debugging is slow over remote connections.

naugehyde, 2025-05-30:
Looks very do-able. 

The protocol is straight GATT.

Apart from battery data, what functionality does the Bluetooth connection provide and which of those will be most useful to you?

I will eventually need remote access to a system with a Litime (or Redodo) battery to vet and complete the code.

Also, do you know if LiTime/Redodo/etc's MPPT and BMS controllers are Bluetooth enabled?

guthip, 2025-06-01:
Great!

I would be happy to be able to read SoC, cell voltage, current for a multi battery setup.
Fantastic would be to read all the rest that the example code provides.
Supurb would be if one can switch on/off the battery output.

Maybe the battery bank configuration should be handled in e.g. node-red.

Should be able to help on access to a Litime battery setup.

Don't know anything about Redodo, Litime/Redodo MPPT.

[Screen Shot 2025-06-01 at 11.34.16 AM.jpg.pdf](https://github.com/user-attachments/files/20541912/Screen.Shot.2025-06-01.at.11.34.16.AM.jpg.pdf)

closed 2025-06-29 by naugehyde