naugehyde/bt-sensors-plugin-sk #48: Fails to update on Signal K Server version 2.14.0

dnoiseux, 2025-06-16:
Jun 16 19:40:14 POST /skServer/appstore/install/bt-sensors-plugin-sk/1.2.1 200 6.376 ms - 36
Jun 16 19:40:14 stderr: /usr/lib/node_modules/npm/node_modules/cacache/lib/verify.js:11 const pMap = require('p-map') ^ Error [ERR_REQUIRE_ESM]: require() of ES Module /usr/lib/node_modules/npm/node_modules/cacache/node_modules/p-map/index.js from /usr/lib/node_modules/npm/node_modules/cacache/lib/verify.js not supported. Instead change the require of index.js in /usr/lib/node_modules/npm/node_modules/cacache/lib/verify.js to a dynamic import() which is available in all CommonJS modules. at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/cacache/lib/verify.js:11:14) at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/cacache/lib/index.js:6:16) at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/make-fetch-happen/lib/cache/entry.js:4:17) at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/make-fetch-happen/lib/cache/index.js:2:20) at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/make-fetch-happen/lib/fetch.js:7:15) at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/make-fetch-happen/lib/index.js:4:15) at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-registry-fetch/lib/index.js:6:15) at Object.<anonymous> (/usr/lib/node_modules/npm/lib/utils/replace-info.js:1:22) at Object.<anonymous> (/usr/lib/node_modules/npm/lib/utils/error-message.js:4:21) at Object.<anonymous> (/usr/lib/node_modules/npm/lib/utils/exit-handler.js:5:22) at module.exports (/usr/lib/node_modules/npm/lib/cli-entry.js:15:23) at module.exports (/usr/lib/node_modules/npm/lib/es6/validate-engines.js:39:10) at module.exports (/usr/lib/node_modules/npm/lib/cli.js:4:31) at Object.<anonymous> (/usr/lib/node_modules/npm/bin/npm-cli.js:2:25) { code: 'ERR_REQUIRE_ESM' } Node.js v20.5.1

naugehyde, 2025-06-18:
Did you solve the issue? Do I need to keep this open?

dnoiseux, 2025-06-26:
I think I just got everything working tonight but I can't even retrace I the step I did.

naugehyde, 2025-06-26:
I'll close the issue.

Bon vent!