naugehyde/bt-sensors-plugin-sk #50: Missing cell voltage values for Victron Lithium Battery Smart

svlibertine, 2025-06-17:
As of 1.2.0 cell voltage values from my Victron Lithium Battery Smart are no longer configurable.

Only the following values are configurable:

RSSI
BMS Flags
Smart Lithium Errors Flags
batteryVoltage
balancer status
batteryTemp

A voltage value for each of the cells in the battery was previously configurable.

SignalK version: 2.14.0.

naugehyde, 2025-06-17:
Apologies. I didn't see that in my (limited) testing.

Please turn on debug and let me know if you see any relevant errors in the
server log and if you do, post them here.

svlibertine, 2025-06-17:
Awesome, thank you so much for the quick response and investigation. Happy to wait for next release.

svlibertine, 2025-06-17:
Haha! Having voltages for the 40 cells in my bank available in SignalK (visualised in KIP with diff/min/max across the whole bank) is extremely helpful compared to connecting to each battery one at a time with VictronConnect. So I am the one who should be thankful :) The UI improvements in Plugin Config in 1.2.0 are great btw!

naugehyde, 2025-06-17:
Wow. You have five batteries in your bank?

svlibertine, 2025-06-17:
> naugehyde left a comment [(naugehyde/bt-sensors-plugin-sk#50)](https://github.com/naugehyde/bt-sensors-plugin-sk/issues/50#issuecomment-2981060888)
> If a Remoran Bluetooth connection to SK is something you're interested in,
> let me know.
>
> The charger doc says they report over Bluetooth.

That would be very interesting indeed!!

The chargers do report over bluetooth, but the app is very limiting. Connecting to SK would be fantastic.

I’m unfortunate not well versed in bluetooth so I would need some pointers as to how and what I can capture from the chargers and provide.

svlibertine, 2025-06-17:
Great, thanks! I will give it a go.

svlibertine, 2025-06-18:
Below is a capture from one of the Remoran devices:

`10:25:04.6610: connected in underlying BLE layer.
10:25:05.1860: discovered services: (
    "81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B"
)
10:25:05.3670: discovered characteristics for service 81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: (
    "62C91222-FAFE-4F6E-95F0-AFC02BD19F2E",
    "F5D12D34-4390-486C-B906-24EA8906AF71",
    "F12A8E25-59F7-42F2-B7AE-BA96FB25C13C"
)
10:25:05.3700: state changed to 'Connected'.
10:25:15.2330: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 a296ce41 00000000 00000000>
10:25:15.2920: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 0ce59441 91000000 e44ae52f 00000000>
10:25:15.3530: received update from characteristic F12A8E25-59F7-42F2-B7AE-BA96FB25C13C___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <8459562e 8459562e 06000100 23000000>
10:25:15.4130: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 a296ce41 00000000 00000000>
10:25:15.4730: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 0ce59441 91000000 e44ae52f 00000000>
10:25:15.5340: received update from characteristic F12A8E25-59F7-42F2-B7AE-BA96FB25C13C___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <8459562e 8459562e 06000100 23000000>
10:25:21.5330: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 1a92ce41 00000000 00000000>
10:25:47.7850: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 bf019541 b2000000 e44ae52f 00000000>
10:25:49.4930: received update from characteristic F12A8E25-59F7-42F2-B7AE-BA96FB25C13C___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <98a6672e 9ba6672e 06000200 45000000>
10:25:55.6440: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 21a5ce41 00000000 00000000>
10:25:58.6750: updated subscription state for characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: subscribed
10:25:59.6670: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 419ace41 00000000 00000000>
10:26:00.1130: updated subscription state for characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: subscribed
10:26:00.6230: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 9d93ce41 00000000 00000000>
10:26:00.6240: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 152c9541 bf000000 e44ae52f 00000000>
10:26:01.5830: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 c49bce41 00000000 00000000>
10:26:01.5840: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 f7129541 c0000000 e44ae52f 00000000>
10:26:02.0930: updated subscription state for characteristic F12A8E25-59F7-42F2-B7AE-BA96FB25C13C___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: subscribed
10:26:02.5440: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 b9b2ce41 00000000 00000000>
10:26:02.5460: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 70fd9441 c1000000 e44ae52f 00000000>
10:26:02.5750: received update from characteristic F12A8E25-59F7-42F2-B7AE-BA96FB25C13C___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <4ec6732e 51c6732e 06000200 53000000>
10:26:03.6540: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 c49bce41 00000000 00000000>
10:26:03.6550: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 7e289541 c2000000 e44ae52f 00000000>
10:26:03.6870: received update from characteristic F12A8E25-59F7-42F2-B7AE-BA96FB25C13C___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <be3b742e c13b742e 06000200 54000000>
10:26:04.6140: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 09a6ce41 00000000 00000000>
10:26:04.6160: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 f7129541 c3000000 e44ae52f 00000000>
10:26:04.6460: received update from characteristic F12A8E25-59F7-42F2-B7AE-BA96FB25C13C___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <d715752e da15752e 06000200 55000000>
10:26:05.5740: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 d3a4ce41 00000000 00000000>
10:26:05.5750: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 00fc9441 c4000000 e44ae52f 00000000>
[... cut ...]
10:26:38.5460: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 0d339641 e5000000 e44ae52f 00000000>
10:26:39.6550: received update from characteristic 62C91222-FAFE-4F6E-95F0-AFC02BD19F2E___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <03000000 00000000 91aace41 00000000 00000000>
10:26:39.6560: received update from characteristic F5D12D34-4390-486C-B906-24EA8906AF71___81D08DF0-C0F8-422A-9D9D-E4379BB1EA3B: <83000000 414f9541 e6000000 e44ae52f 00000000>
10:26:40.3680: state changed to 'Disconnected'.
User initiated device disconnection.
10:26:40.3710: disconnected in underlying BLE layer.`

svlibertine, 2025-06-18:
This is the data available in the native app:

[image]

svlibertine, 2025-06-18:
Device shows up in Plugin Config:

[image]

svlibertine, 2025-06-18:
[output.txt](https://github.com/user-attachments/files/20800489/output.txt)

naugehyde, 2025-06-18:
I've created a new issue. Let's move the conversation there.

closed 2025-06-18 by naugehyde