naugehyde/bt-sensors-plugin-sk #56: victron BMV712 . no aux field

gildas-declercq, 2025-06-30:
according to 

[image]

I see like a dropdown list but with nothing else in the list

[image]

naugehyde, 2025-06-30:
> the complete list of fields
>
> image.png (view on web)

I see a starter battery voltage path at the bottom. Isn't that what you're
looking for?

gildas-declercq, 2025-06-30:
could be but does not show in SK

[image]

gildas-declercq, 2025-06-30:
let me now how you want to connect

naugehyde, 2025-06-30:
any ssh connection will do.

just need credentials for your server and its public address or a vpn that
you can share with me (Zerotier, Tailscale etc)

I also need admin rights on SK.

gildas-declercq, 2025-06-30:
ok we will have to organise that.  if you know what I can add on openplotter so you can access , as it is not on public address.

won't have time right now . it is 6pm here and week should be pretty busy

gildas-declercq, 2025-07-05:
Hi Andy, sorry for the delay. here are the details of my last tests.
fresh start of the pi.

[image]

only smart solar data show up

[image]

which does make sense, yesterday I deleted the BMV from config

[image]

gildas-declercq, 2025-07-05:
but I don't see the BMV anymore

[image]

naugehyde, 2025-07-05:
> but I don't see the BMV anymore
> image.png (view on web)

Try executing

``sudo service bluetooth restart``

from a command line

If you still don't see the devices, there may be radio interference.

gildas-declercq, 2025-07-06:
restart service done
had to restart sk as nothing coming from bt sensor plugin
i see a new version of the plugin, shoud i updtate ?
I still dont see the BMV.
I do experience some connexion issues though.  on my tablet with victron connect I have to refresh sometime, and I did not realize that before.
I will come back to the boat, in about 8hours.

gildas-declercq, 2025-07-06:
ooops. the scan was at zero

[image]
i will reconfigure those

gildas-declercq, 2025-07-06:
I sent you what you requested by mail hope this can help
let me know if we have to organize a vpn access to the rpi.  I will have to be on my boat.
I am in GMT +11