naugehyde/bt-sensors-plugin-sk #57: Govee Thermometer showing wrong values

benschmi, 2025-07-01:
Hi, first of all thank you very much for all your efforts to the project! I love OpenPlotter and I love the community around it :)

I'm integrating a Govee H5075 to measure the temperature of my fridge. Integration worked very smoothly, but OpenPlotter is showing weird values. For temperature, they are laying somewhere between 750 and 960 and for humidity I get something around 2.5. They also vary greatly, whereas the on-device display and also the values in the Govee app are absolutely correct. 

Do you know what the issue could be?

benschmi, 2025-07-03:
So this is basically what bluetoothctl returns for my device:

[CHG] Device A4:C1:38:D1:EC:6B ManufacturerData Key: 0xec88
[CHG] Device A4:C1:38:D1:EC:6B ManufacturerData Value:
[CHG] Device A4:C1:38:D1:EC:6B ManufacturerData Key: 0x004c
[CHG] Device A4:C1:38:D1:EC:6B ManufacturerData Value:
[CHG] Device A4:C1:38:D1:EC:6B RSSI: -46

naugehyde, 2025-07-03:
Got to it last night. I can send you a file to add to your system for
testing or you can wait until the next release to test.

Let me know what you'd prefer.

For future reference, the value on the other end of ``[CHG] Device
A4:C1:38:D1:EC:6B ManufacturerData Value:`` is missing and is what I parse
to get the device's data. It appears below the line you captured and will
not get picked up by a command like ``bluetoothctl | grep <mac_address>``

If you could provide those values in the next few days, it'd be helpful for
testing my assumptions about the parsing that's in the existing code.

Herz-aus-Stahl, 2025-07-03:
i have the same problem with the Xiaomi Thermometers. with the old plugin 1.1.2 it works, with 1.2.3 the values are too low, instead of 23.8 it shows 2.4.

naugehyde, 2025-07-03:
That's a different sensor class but it's still a problem!

Please open a new issue with the model of the Xiaomi Thermometer in the
description. Please indicate which firmware (stock Xiaomi or ATC) you are
using.

benschmi, 2025-07-04:
Found it, hope that's it:
```
[CHG] Device A4:C1:38:D1:EC:6B RSSI: -47
[CHG] Device A4:C1:38:D1:EC:6B ManufacturerData Key: 0xec88
[CHG] Device A4:C1:38:D1:EC:6B ManufacturerData Value:
  00 03 bb 94 64 00                                ..?.d.          
[CHG] Device A4:C1:38:D1:EC:6B ManufacturerData Key: 0x004c
[CHG] Device A4:C1:38:D1:EC:6B ManufacturerData Value:
  02 15 49 4e 54 45 4c 4c 49 5f 52 4f 43 4b 53 5f  ..INTELLI_ROCKS_
  48 57 50 75 f2 ff 0c                             HWPu??.    
```

I can test the file earliest Tuesday.

naugehyde, 2025-07-04:
That data shows 24.5c and 62.8% humidity and 100% battery. Is that
correct-ish?

naugehyde, 2025-07-04:
> Yes, that seems about right.

Hurrah! The parsing is very strange for the Govee 5075. For some reason
they pack signed temperature and humidity into a 24 bit unsigned integer.
But I think it will work now. Expect a file or a release notice in the next
few days.