naugehyde/bt-sensors-plugin-sk #60: Recent releases only report Govee signal strength

Peter-Petrik, 2025-07-07:
Same Govee sensors as used in #5. With recent releases (1.2.4?) the bluetooth connection is made, signal is reported but no temperature, humidity, or battery status.

Has there been any change to needed configuration to make these work again?

naugehyde, 2025-07-07:
Hi Peter. It's possible there are issues here as I had to move some of the
code around for the Govee sensors.

Which model of sensors are you using?

If you're not seeing data, try deleting the configuration for the sensor
and re-adding it.

naugehyde, 2025-07-07:
> Which model of sensors are you using?
>
> Govee GVH5100
>
> If you're not seeing data, try deleting the configuration for the sensor
> and re-adding it.
>
> Deleted, added back in, and still only seeing the signal and no other data
> points.

I'll look into it ASAP

naugehyde, 2025-07-07:
Please attach the config file for the plugin. Thank you.

Peter-Petrik, 2025-07-07:
> Please attach the config file for the plugin

Sent via email. Thank you!

closed 2025-07-07 by naugehyde

naugehyde, 2025-07-07:
> That resolved it. Thank you for the prompt response!

I hate bugs!