naugehyde/bt-sensors-plugin-sk #64: 2.1.5 installation problem

ckcrs, 2025-07-15:
Hello, thanks for the plug-in, I'm planning to use it with a Calypso Ultrasonic wind sensor in near future, trying to install but I cannot.

1- Signal K 2.14.4, node -v && npm -v shows v20.19.1 and 11.4.2 on my pi Zero 2 W, bullseye 32:  Linux bullseye32 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l

2- sudo apt update says nothing to update.

3- I've tried App Store install with result "failed" or

"$ npm i bt-sensors-plugin-sk" command without success:
pi@bullseye32:~ $ npm i bt-sensors-plugin-sk 
npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm warn deprecated har-validator@5.1.5: this library is no longer supported
npm warn deprecated npmlog@4.1.2: This package is no longer supported.
npm warn deprecated rimraf@3.0.2: Rimraf versions prior to v4 are no longer supported
npm warn deprecated are-we-there-yet@1.1.7: This package is no longer supported.
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm warn deprecated gauge@2.7.4: This package is no longer supported.
npm warn deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
⠇
<--- Last few GCs --->

[11093:0x4e0c4d0]   229355 ms: Scavenge 121.0 (128.9) -> 120.7 (128.9) MB, 3.06 / 0.00 ms  (average mu = 0.741, current mu = 0.602) task; 
[11093:0x4e0c4d0]   229443 ms: Scavenge 121.1 (128.9) -> 120.7 (129.4) MB, 33.09 / 0.00 ms  (average mu = 0.741, current mu = 0.602) allocation failure; 
[11093:0x4e0c4d0]   232223 ms: Mark-Compact 121.2 (129.4) -> 120.5 (129.6) MB, 2674.99 / 0.00 ms  (average mu = 0.612, current mu = 0.517) allocation failure; scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: Reached heap limit Allocation failed - JavaScript heap out of memory
----- Native stack trace -----

Aborted

4- Can you elaborate your note of "Submit the main configuration then enable and optionally disable Debug" please, as all the plugins lost and a message on "something went wrong" message displayed on server plugin config screen, please. 

5- Or my pi Zero 2 W ver 1; is not powerful enough to run?

Thanks

naugehyde, 2025-07-15:
1.2.5 solves the initial install refresh problem so ignore #4

Your Zero 2W may be powerful enough to run SK and the plugin but it looks like it doesn't have enough RAM to install the package. You can try creating a swap space on disk and re-running but the Zero2 isn't ideal for running SK. 1GB doesn't seem to be enough these days for Node 22 and later.

You may also encounter signal strength issues and may not be able to connect or stay connected to the Calypso from the Zero2 if it's atop your mast. I guess you'll let me know -- assuming you can solve the available RAM issues. 

1) Reconsider the Zero2 as a SK server. Look at the 4B or 5.

2) If you don't want to go the hardware upgrade route and if you're running a desktop on the Zero2 I'd recommend running a headless server (no desktop).

3) If you stick with the Zero2 I'd recommend creating or enlarging the swap space on disk.

4) no matter what you may need a more powerful Bluetooth device (see readme)

naugehyde, 2025-07-16:
I installed the latest SK release and the BT plugin on a Zero2W running headless RPiOS (bookworm 64bit) and node 22 with a disk swap size of 512mb. 

It wasn't fast but it installed and worked perfectly sell. 

The errors you were getting are likely a result of memory limitations encountered during compilation of bindings for dbus-next or some other dependent package. The error only incidentally occurred while installing the plugin. It looks like memory issues were just waiting to happen

So...

If you have a desktop installed on the pi, the first thing I would do is remove  desktop from the installation and think about other ways of interacting with your Pi. 

Then I'd try stopping SK and running the install without SK in memory. If that doesn't work, I'd increase the swap size to 512mb (I'd do that anyway come to think of it). 

Then, if you still can't install the plugin, I'd start my install from scratch, installing bookworm 64 bit (no desktop) and a swap disk of 512mb or greater. 

If none of that works I'd get a Pi 5 (2Gb or more if headless, 4g or more if running a desktop), and then prepare to set sail on my new Pi/SK-venture.

I can tell you that the Calypso Ultrasonic works very nicely on my boat but you will need a more powerful BT receiver than the onboard one provided by the pi-- especially if the calypso is atop the mast. I recommend the TP500 Plus.

Good luck and let me/us know how it goes.

ckcrs, 2025-07-17:
Thank you very much for detailed reply. Here what I've tried just now: ( with node version  v20.19.1)
1-Stop VNC service, config pi to boot to console with auto login.
2-Check mem available round 256 MB
3- try to run SignalK plugin loading, shows failed, it shows the plugin on plugin config window but when I try to configure, it says "something went wrong"
4- No stopped signal service and socket and trying "npm i bt-sensors-plugin-sk" it also gave fatal error : "FATAL ERROR: Reached heap limit Allocation failed - JavaScript heap out of memory"
5- added a swap file of size 512 in total 600 MB swap size in two files as:
pi@bullseye32:~ $ free -h
               total        used        free      shared  buff/cache   available
Mem:           474Mi       182Mi        92Mi       0.0Ki       198Mi       237Mi
Swap:          611Mi       1.0Mi       610Mi
6-uninstall plugin ( corrupted installation) via SignalK appstore, reboot and then try reinstall via App Store once again, result is failed installation and plugin config screen gives "something went wrong" again, now uninstalled it via App Store, try installing again with npm but:
-7 stopped signal and try installing via npm,
npm error code ENOTEMPTY
npm error syscall rename
npm error path /home/pi/node_modules/abbrev
npm error dest /home/pi/node_modules/.abbrev-8GDpPNSr
npm error errno -39
npm error ENOTEMPTY: directory not empty, rename '/home/pi/node_modules/abbrev' -> '/home/pi/node_modules/.abbrev-8GDpPNSr'
npm error A complete log of this run can be found in: /home/pi/.npm/_logs/2025-07-17T08_35_32_614Z-debug-0.log
-8 reboot pi and try again npm with SignalK stopped now, result is same??
-9 stopped here as I don't want to continue further pushing piZero, I will try to use an old "calypso" plugin I had installed.
-10 My conclusion is piZero is not powerful enough  :( Maybe in the future give another try with Pi 4.
Thanks and kind regards

closed 2025-07-17 by ckcrs

naugehyde, 2025-07-17:
1) where is you signalk home (location of settings.json etc)? is it
/home/pi/.signalk or /home/pi? The error is occurring at
/home/pi/node_modules with a module (abbrev) that isn't a dependent on my
plugin. ``npm error ENOTEMPTY: directory not empty, rename
'/home/pi/node_modules/abbrev' ->
'/home/pi/node_modules/.abbrev-8GDpPNSr'`` indicates an abandoned install
at the /home/pi level not the /home/pi/.signalk level (default install
location)

2) the "something went wrong" message occurs when any error is found in
loading the configuration panel. If the bt-sensors plugin is not loaded, a
different plugin is corrupted. To investigate on Chrome open up the
developer tools option and look at the console for errors.

3) wherever your signalk home is (see #1 above) you may have a corrupted
node_modules directory that needs to be rebuilt.

4) Here is my Zero2W's memory:
``
free -h
               total        used        free      shared  buff/cache
available
Mem:           416Mi       146Mi       113Mi       4.7Mi       216Mi
269Mi
Swap:          511Mi        31Mi       480Mi``

Which is sufficient for installing and running the plugin.

5) The major difference between your setup and mine is I'm running
64bit RPI OS.

6) I wouldn't give up. If only because it looks like you're close from over
here (even if you're getting frustrating errors). You may need to reinstall
SK or at least reinstall the plugin from the correct signalk home.

ckcrs, 2025-07-21:
Thanks for your kind support.
1) /home/pi/.signalk contains settings.json whereas /home/pi contains .signalk/ and others . And: /home/pi/node_modules/bt-sensors-plugin-sk
2) Retried installing version 1.2.15.1 via web apps it said failed but restarted signalk, I can see it in plugin configuration screen as enabled, but when I try to submit "something went wrong" appeared again.
3)I will make a clean pi image and signalk installation probably in winter, better to sail now.
Thanks and fair winds...