naugehyde/bt-sensors-plugin-sk #65: Victron device identification

Peter-Petrik, 2025-07-16:
I have 3 identical devices: Victron SmartSolar MPPT 100|20 48V

One is identified as such, the other two just show `Victron VictronSolarCharger` and different Model ID. From usability standpoint and reporting I see no difference between the devices - the issue appears to be purely cosmetic.

[image]

naugehyde, 2025-07-16:
I saw that recently on my boat (my bt receiver picks up my neighboring boat's Solar Charger.) I'll look into it.

naugehyde, 2025-07-17:
Is VE Smart Networking on for any of the devices? 

There is an anomaly that can occur with devices that have VE Smart Networking on that can result in the model ID not being determined and the results you are seeing.

naugehyde, 2025-07-19:
Check out version 1.2.5-1. 

You will need to save your Victron configs but the naming should be consistent after that.

closed 2025-07-19 by naugehyde

Peter-Petrik, 2025-07-19:
Worked great right after restart! Thank you!