naugehyde/bt-sensors-plugin-sk #68: Manual Setting sensor class doesn't work

timcade75, 2025-07-20:
Hello, I am in the process of trying to add my lossigy batteries (made by Shenzhen) to signal k. When I select ShenzhenLIONBMS or anything else for that matter, then save, the device shows up in configured but the sensor class clears out and it is just blank.   I have tried to classify it as other BMS listed as well. Nothing seems to stick.

I am certain I am missing a step somewhere in the documentation.  Can someone point me in the right direction?

naugehyde, 2025-07-21:
The onboard BMS for the Lossigy appears to be a Jibaida or JBD BMS. I can help you confirm this if you send me the basic information page from the Lossigy app. To confirm, check if the device model on the basic information page begins with the letters JBD. 

If so, use the JBDBMS class instead of the ShenzhenLiOn class.

timcade75, 2025-07-21:
got it! so, it is a JBD model DP24S002. 
still getting the same result.  

log shows a timeout. see below

Jul 21 13:49:20 GET /signalk/v1/api/vessels/ 200 1.031 ms - 404
Jul 21 13:49:20 GET /signalk/v1/api/atons/ 404 0.700 ms - 160
Jul 21 13:49:28 2025-07-21T18:49:28.024Z bt-sensors-plugin-sk { active: true, discoveryTimeout: 30, params: { name: 'STARBOARD', sensorClass: 'JBDBMS' }, paths: { RSSI: 'sensors.{macAndName}.RSSI' }, mac_address: 'A4:C1:37:33:46:9E' }
Jul 21 13:49:28 2025-07-21T18:49:28.025Z bt-sensors-plugin-sk Plugin options saved
Jul 21 13:49:28 POST /plugins/bt-sensors-plugin-sk/updateSensorData 200 2.415 ms - 28
Jul 21 13:49:30 GET /signalk/v1/api/vessels/ 200 0.722 ms - 404
Jul 21 13:49:30 GET /signalk/v1/api/atons/ 404 0.680 ms - 160
Jul 21 13:49:31 2025-07-21T18:49:31.351Z bt-sensors-plugin-sk { adapter: 'hci0', transport: 'le', duplicateData: false, discoveryTimeout: 30, discoveryInterval: 10 }

naugehyde, 2025-07-21:
Hmm. That's not looking right.

When you select the Device in the config do you see a series of paths (voltage etc.?)

If not, refresh the browser page and let me know if you do.

naugehyde, 2025-07-21:
Can you send a screenshot of the config and your plugin config file (~/.signalk/plugin-config-data/bt-sensors-plugin-sk.json) ?

Thank you.

timcade75, 2025-07-21:
that is what happens after I click save

naugehyde, 2025-07-21:
The problem is on my end (the plugin is storing the sensor class as Unknown and not as JBDBMS).  I thought I had a workaround but it doesn't seem to be working for you. I fixed it in the latest commits but they're not in the app store yet. 

In the meantime, edit the config file directly. Change

`"sensorClass": "UNKNOWN" `to

`"sensorClass": "JBDBMS"` and

`"discoveryTimeout": 30,` to

`"discoveryTimeout": 90,`

Then restart Signalk.  And then, select the device and see if the paths are visible. If they are, save the configuration.

I have to hit the road but we can pick this up tomorrow if necessary.

timcade75, 2025-07-21:
thank you so much for all the attention.  I will let you know how it turns out either way

timcade75, 2025-07-21:
This worked. I was able to add all my batteries.

naugehyde, 2025-07-21:
> Am I only allowed two batteries? my boat is full electric and I have 4. it
> can't seem to create the third so far.

There's no limitation that the plug-in imposes. Sounds like a connection
problem. Check your debug logs for any connection errors.

If your devices are out of range, read the README about how to improve your
range.