naugehyde/bt-sensors-plugin-sk #71: New device: Eco-worthy LiFePo4 batteries BT controlled

lula6894, 2025-08-09:
Hi Naugehyde,

Thanks again for this awesome plugin! I’m using it as the heart of my boat’s sensor setup, and it’s been working brilliantly.
You might remember me—we chatted a while back about adding support for a Lancol voltage meter. Well, I’ve now switched from lead-acid to LiFePO₄ batteries, and these new ones have Bluetooth control. My first thought was to check if the plugin already supports Eco-Worthy batteries.
Do you think it might be possible to add them to your list of supported devices in the future?
Big thanks in advance, and hats off for the amazing work you’ve done with this plugin!

All the best,

naugehyde, 2025-08-10:
There's a utility program I wrote called ``connectUUID.exp`` that's in the main codebase here in github. 

The UUID in this case appears to 0xffE1 

1) Copy connectUUID.exp into a working directory on your signalk machine. 
2) install expect utility by entering ``sudo apt install expect``
3) execute the utility like so: 
``./connectUUID.exp <mac_address> <UUID> <number of data packets>``
4) output the result to a file and attach it here.

I hope this works. I'm not going to be able to walk you through the process of getting it to work so please consult google and or ChatGPT first.

Thanks!

lula6894, 2025-08-11:
I´m running a PI5 8GB NVME disk with bookworm and XDE Plasma, SignalK and OpenCPN.
I disconnected de BT Mouse and ran again. This is the file:

[eco_worthy_ble_2025-08-11_2127.log](https://github.com/user-attachments/files/21723484/eco_worthy_ble_2025-08-11_2127.log)

naugehyde, 2025-08-11:
THat's a little better.

It's not finding the device which means you're probably connected to it on your phone. 

Be sure to disconnect on phone then re-run script.

naugehyde, 2025-08-11:
What is the mac address you're providing to the script?

lula6894, 2025-08-12:
./connectUUID.exp A5:C2:37:46:2D:99 0xFFE1 200 | tee eco_worthy_ble_$(date +%F_%H%M).log

I can provide you ssh access, provide me an email and I`ll send the grants.

naugehyde, 2025-08-12:
DM me on discord if you can.

If not email andy [at] longslowno dotcom

lula6894, 2025-08-12:
sent

lula6894, 2025-08-12:
As I`m sailing and i`m connecting through a 4G router if you cannot access is because I have no coverage

naugehyde, 2025-08-17:
check out latest beta on npm and latest commit here on git for EcoWorthy support.

it looks like you will need the TP UB 500 Plus to get and maintain a connection.

lula6894, 2025-08-17:
Thanks!!! Just back from sailing today and the TP USB 500 was waiting at home. I'll test tomorrow in the boat. Keep you updated if you want to login and check.
All the best

Best
-Fernando

naugehyde, 2025-08-17:
Very unlikely the sensor class works at all on first try buy I did do some
rudimentary testing on my boat with a fake software device.

closed 2025-08-21 by naugehyde