naugehyde/bt-sensors-plugin-sk #72: Device configs no saving sensor class

Mechcondrid, 2025-08-10:
after updating the plugin and signalk, i can no longer add new devices and have them populate correctly.

i am running on a hatlabs halpi using a CM4 module.

i can watch the devices scan, populate, can pick the class, but when i hit save, the config file is saved but the sensor class is left undefined and no paths are added.

naugehyde, 2025-08-10:
It sounds like a duplicate of https://github.com/naugehyde/bt-sensors-plugin-sk/issues/68.

The fix is in the repository and the beta on npm. If you don't want to update to a beta there is a simple hack mentioned in the discussion above (manually edit the config sensorClass).

Mechcondrid, 2025-08-10:
yeah, actually thought of that before i even went to GH, no go, it ignores the class in the file and never adds any paths, tried reboots, enable/disable cycles, nada.

naugehyde, 2025-08-10:
Also turn on debug for the plugin and let me know if there are any errors.

If it's not a duplicate of #68 something else is going wrong!

naugehyde, 2025-08-10:
1) turn off plugin

naugehyde, 2025-08-10:
Also,

1) Are you running a desktop on the server? 
2) Are you using the server as a WiFi access point?

Mechcondrid, 2025-08-11:
so yeah, now almost 10pm, still no drop in the connection. i don't think it's about connecting and holding it, as i mentioned that was working BEAUTIFULLY.

[image]

naugehyde, 2025-08-11:
Something else is going wrong then.

Are you certain the battery's BMS is from Jibaida (JBD)?

What's the brand and model of the battery? Did it come with an app? If so, what one?

Mechcondrid, 2025-08-11:
sent you a PM

naugehyde, 2025-08-21:
@Mechcondrid can you confirm that after some testing and experimenting it looks like the problem of connecting the plugin to to the JBDBMS device was two-fold:

1) The lxpanel bluetooth was interfering with the BT adapter state
2) The JBDBMS BT connection needs about a minute to recycle after disconnection from any source.

Thanks.

closed 2025-08-21 by naugehyde