naugehyde/bt-sensors-plugin-sk #73: Govee H5074 only shows RSSI

CaptIgmu, 2025-08-11:
Temp, humidity, battery.strength not showing. No paths created regardless of how configured on SK, using RPi5.

naugehyde, 2025-08-11:
Try deleting the config then recreating it.

naugehyde, 2025-08-22:
Weird. I'm running H5074s without a problem.

If you have the time, can you send pic of the device and your config file and turn on debug and send any errors that you see re: communicating with the device?

It will be helpful to know if there are any incompatibilities in the Govee firmware or in my plugin that account for the behavior you're seeing.

CaptIgmu, 2025-08-27:
Voila! The RPi5 on my boat works fine with the Govee H5074. Also, my Ruuvi sensor is working with this plugin. My RPi5 at home, used for testing/development, is surrounded by a large array of bluetooth devices and the log shows some time outs. I'm just happy it works on the boat. Thanks for this plugin!

naugehyde, 2025-08-27:
Interesting. And glad to hear!

The two big culprits with interference that result in timeouts and packets not being received are poorly shielded USB-3 and HDMI both of which operate at approximately 2x the transmission rate of BT-LE which can cause havoc. 

If your home device had an active HDMI or USB-3 connection that might explain the signal issues.