naugehyde/bt-sensors-plugin-sk #75: Power cycle BT if no data is coming in

KEGustafsson, 2025-08-17:
Would it be possible to build healthy check if there is no BT data coming in for certain period then BT module would be power cycled?
Either using `node-ble` or `bluetoothctl`

Time to time data stops coming in and `bluetoothct` and `power off` -> `power on` -> `quit`, recover situation.
Same should be doable using `node-ble` if I have understood correctly.

KEGustafsson, 2025-08-20:
This is how this looks when bt data stops. Single readings in between is from SK restart, but cannot say are value from some buffer or real time values.

[image]

```
node@xavier:~/.signalk$ bluetoothctl show
Controller D8:C0:A6:26:1C:3E (public)
        Name: xavier
        Alias: xavier
        Class: 0x00000000 (0)
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4 (180)
        Pairable: no
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0535
        Discovering: yes
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x04 (4)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
        SupportedSecondaryChannels: 1M
        SupportedSecondaryChannels: 2M
        SupportedSecondaryChannels: Coded
```
```
node@xavier:~/.signalk$ hciconfig hci0
hci0:   Type: Primary  Bus: USB
        BD Address: D8:C0:A6:26:1C:3E  ACL MTU: 1021:6  SCO MTU: 255:12
        UP RUNNING
        RX bytes:43987583 acl:0 sco:0 events:662784 errors:0
        TX bytes:1226624 acl:0 sco:0 commands:128923 errors:0
```
```
node@xavier:~/.signalk$ timeout 60 bluetoothctl
hci0 new_settings: powered bondable ssp br/edr le secure-conn
Agent registered
[CHG] Controller D8:C0:A6:26:1C:3E Pairable: yes
[bluetooth]#
node@xavier:~/.signalk$
```

Then  `bluetoothctl` -> `power off` -> `power on` and we are live again.
```
node@xavier:~/.signalk$ bluetoothctl
hci0 new_settings: powered bondable ssp br/edr le secure-conn
Agent registered
[CHG] Controller D8:C0:A6:26:1C:3E Pairable: yes
```
```
[bluetooth]# power off
hci0 new_settings: bondable ssp br/edr le secure-conn
Changing power off succeeded
[CHG] Controller D8:C0:A6:26:1C:3E Powered: no
[CHG] Controller D8:C0:A6:26:1C:3E Discovering: no
```
```
[bluetooth]# power on
hci0 new_settings: powered bondable ssp br/edr le secure-conn
Changing power on succeeded
[CHG] Controller D8:C0:A6:26:1C:3E Powered: yes
[CHG] Controller D8:C0:A6:26:1C:3E Discovering: yes
[NEW] Device 20:80:A8:C0:44:D0 20-80-A8-C0-44-D0
[NEW] Device C0:8F:AE:FC:FB:91 Arabella DCDC charge
[NEW] Device C6:06:F4:36:0A:7C Arabella main charge
[NEW] Device AC:AC:AC:88:2C:64 AC-AC-AC-88-2C-64
[NEW] Device C3:20:FC:31:7B:27 Ruuvi 7B27
[NEW] Device E9:02:0B:2E:28:13 SmartSolar HQ2135P2JEW
[NEW] Device D4:28:F4:68:E8:10 D4-28-F4-68-E8-10
[NEW] Device D2:3E:79:F3:A4:82 Ruuvi A482
[NEW] Device 74:5C:9C:64:28:A0 74-5C-9C-64-28-A0
[NEW] Device A8:2C:FC:14:B4:50 A8-2C-FC-14-B4-50
[NEW] Device C4:33:D2:FC:B8:75 Ruuvi B875
[NEW] Device 68:60:98:10:DC:EC 68-60-98-10-DC-EC
[NEW] Device 7C:14:50:0C:5C:20 7C-14-50-0C-5C-20
[CHG] Device D4:28:F4:68:E8:10 RSSI: 0xffffff9d (-99)
[CHG] Device 74:5C:9C:64:28:A0 RSSI: 0xffffff9b (-101)
[CHG] Device 7C:14:50:0C:5C:20 RSSI: 0xffffffa1 (-95)
[CHG] Device 7C:14:50:0C:5C:20 ManufacturerData.Key: 0x060f (1551)
[CHG] Device 7C:14:50:0C:5C:20 ManufacturerData.Value:
  00 17 88 01 07 2f 07 7a 45 61 73 79 41 69 72 20  ...../.zEasyAir
  53 4e 48 32 31 30 20 4d 43 00 00                 SNH210 MC..
[NEW] Device DA:B4:85:6A:76:3A DA-B4-85-6A-76-3A
[NEW] Device F3:13:D0:D4:67:F7 Ruuvi 67F7
```

naugehyde, 2025-08-20:
Something turned the scanner off:

` Discoverable: no`

It's possible another app did that. Searching for a device using Blueman and its like could have tried to turn it off (they shouldn't have control over it by design but you never know.)

It's also possible that the system turns off the scanner automatically after some timeout but I've never heard of that before.

It's also possible that Bluez via Dbus ran out of resources for some reason and turned off scanning in the process of recovery. 

Did you get see errors from the plugin? If it was Dbus failing you would have. Note: ordinary errors like "could not communicate are not unusual")

KEGustafsson, 2025-09-07:
> dbus-monitor --system

```
dbus-monitor --system
dbus-monitor: unable to enable new-style monitoring: org.freedesktop.DBus.Error.AccessDenied: "Rejected send message, 1 matched rules; type="method_call", sender=":1.712" (uid=1000 pid=1375719 comm="dbus-monitor --system " label="kernel") interface="org.freedesktop.DBus.Monitoring" member="BecomeMonitor" error name="(unset)" requested_reply="0" destination="org.freedesktop.DBus" (bus)". Falling back to eavesdropping.
signal time=1757243552.915875 sender=org.freedesktop.DBus -> destination=:1.712 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.712"
```
Not activity in `dbus` until `bluetoothctl`-> `power off`-> `power on`.

naugehyde, 2025-09-07:
@KEGustafsson 

I've been experimenting with a fix to dbus-next that really needs to be main-lined that may help prevent the errors you're seeing. The patch is in a forked repo here:

https://github.com/JellyBrick/node-dbus-next/pull/4/commits/e875f304780e0d8108dbe1d7e7bdcf6ba9e9313f

The relevant discussion is here:

https://github.com/JellyBrick/node-dbus-next/pull/4

naugehyde, 2025-09-07:
Also if the situation arises again, can you see if simply restarting the bluetooth service will cure the condition?

KEGustafsson, 2025-09-08:
Is this patch included to test releases or need to be build by myself?

KEGustafsson, 2025-09-08:
Now I have latest beta in testing.

naugehyde, 2025-09-10:
Latest beta (1.2.6-beta-6d) has the fix to DBus-next implemented properly -- whether it makes a difference or not...

naugehyde, 2025-10-16:
There's a Beta available here: https://www.npmjs.com/package/bt-sensors-plugin-sk/v/1.3.4-beta that includes the inactivity timeout option and logic. (https://github.com/naugehyde/bt-sensors-plugin-sk/commit/f5520816013eebe4bd9b86138b5ef59a47eb56a6)

I'll need your help in testing this before releasing. Please install the beta and let it run with Debug on until the problem arises. NOTE: If it's working the only indication you'll get is in the server and system log.

Thank you!

KEGustafsson, 2025-10-16:
Installed 1.3.4-beta. Test going on.

KEGustafsson, 2025-10-17:
Is the plugin indicating if it has powercycled bt?

naugehyde, 2025-10-17:
yes. you'll get a debug message in the log.

naugehyde, 2025-10-17:
Wondering if the design should be:  the plugin raises an alarm which is then handled by a Node-Red flow or other process that then sends a recycle adapter API call to the plugin. Or maybe that's overkill.

KEGustafsson, 2025-10-21:
BT data stopped, but plugin did not noticed that, not either powercycled bt.

Can you pin point monitoring code part?

KEGustafsson, 2025-10-21:
I'll keep this bt issue on my HW and start to check why your code is not catching it. Good thing is that SK restart is not making it go away so I can debug code easily now until I find right method to catch and reset.

naugehyde, 2025-10-21:
Can you confirm if bluetoothctl show is returning 

```Discovering: yes ```

KEGustafsson, 2025-10-21:
Yep
`Discovering: yes`

naugehyde, 2025-10-31:
Try the latest commits and see if it works.

BTSensor::elapsedTimeSinceLastContact() will return Infinity if device is an OutOfRangeDevice.

Logic is: if ALL Sensors have not emitted any changed properties within the inactivity threshold at plugin startup or at any time during plugin run, the plugin will attempt to recycle the BT adapter.

The problem with this logic is: if ALL sensors are active but Out of Range for ordinary reasons (dead sensor battery, sensor removed from boat etc.) the plugin will repeatedly start and power cycle the adapter to no good effect.

KEGustafsson, 2025-11-02:
Very good! I'm still waiting my bt to hangup. Running v1.3.4

naugehyde, 2025-11-02:
FYI I'm using the latest commits -- not yet published.

naugehyde, 2025-11-02:
There's a not yet published change in BTSensor.js that'll make a difference at detecting a non-working BT Adapter at startup (https://github.com/naugehyde/bt-sensors-plugin-sk/commit/fd23a3363c0b6448fd401908143eb79661133613)

closed 2025-11-27 by naugehyde