naugehyde/bt-sensors-plugin-sk #76: ECO Worthy 100A LiFePo

L0wbird81, 2025-08-17:
Hey naugehyde,

i read that a few people are having problems with your plugin. The type cannot be saved for certain Bluetooth devices, and only the RSSI value is output, but no other data is displayed. I have exactly the same problem with my EcoWorthy 100A LiFePo4 battery from EcoWorthy. The BMS is recognized as DP04S007L4S100A, which apparently belongs to jaibaida. I also entered JBDBMS directly into the config, as suggested in another issue. I would like to test your Beta 2, since you explicitly mention the ECO Worthy batteries. However, I'm not sure how to install this beta into my existing AvNAv installation. I would be happy to test the version and support you in this regard.
Best regards from Germany, Chris

naugehyde, 2025-08-17:
Hi Chris,

Yes, that's very likely a JBD device.

To install the beta go to your signalk home (usually ~/.signalk) and type:

`npm i bt-sensors-plugin-sk@1.2.6-beta-2`

then restart SK. 

You'll want to set your BMS up  as a JBDBMS device not an EcoWorthy. But your results will be useful.

FYI There are two problems colliding at the moment. 1) changes to latest RPi Desktop installations that are apparently not cooperating with the plugin. 2) The aforementioned plugin issue where the class selection isn't working. I'm tracking that down. It may be as simple as the plugin failing to report an error connecting to or staying connected to the device for whatever reason.

naugehyde, 2025-08-17:
Also, if there's any chance I could SSH into your system, DM me on Discord. It will expedite fixing this.

In the meantime, turn debug on for the plugin and post any errors you see in the server log here.

L0wbird81, 2025-08-17:
I executed the command in the signalk directory. I'll be back on the boat tomorrow and see if anything works. Thanks for that.

naugehyde, 2025-08-17:
If you're not running node 22 or greater (recommended for SK >= 2.15) you'll need the latest commit to EcoWorthy.js referenced above.

L0wbird81, 2025-08-19:
Hey naugehyde, how can I check wich version of the plugin is installed? I guess the installation of the plugin doesn’t work like suspected. I  also tried to install the newer node on my raspberry, but it is still version 20.*.*, not 22.**
So I will try another attempt to install the Beta 3
Best regards from Germany. 
Chris

naugehyde, 2025-08-19:
The plug-in version is printed in the server log at server startup.

See the signalk README for links to nodejs installation tips.

If the plug-in is up to date you should be able to set up your eco worthy battery as a JBDBMS device. 

If you experience any issues, turn in plug-in debug and send any relevant server log entries to this forum

naugehyde, 2025-08-20:
Gutentag, Chris.

The error loading EcoWorthy stems from running node 20. The published version only works in node > 22. The next beta will support both node versions (though SK will, at start up, warn you about running version 20).

The BadRequest error probably stems from the JBD device being unconnectable for some reason (The plugin needs better error reporting).  There are a number of reasons why you may not be able to connect.

First, test the connection.

from a command line execute:

`bluetoothctl connect <mac address of JBD device>`

and report the results here.

Then report results of: 

`bluetoothctl info <mac address of JBD device>`

Tchuss!

L0wbird81, 2025-08-20:
Hallo naugehyde,

for the first Command the results: (connetion successful)
pi@raspberrypi:~ $ bluetoothctl connect A5:C2:37:36:32:A7
Attempting to connect to A5:C2:37:36:32:A7
[CHG] Device E4:EE:92:E0:6E:03 RSSI: -73
[CHG] Device E4:EE:92:E0:6E:03 ManufacturerData Key: 0x02e1
[CHG] Device E4:EE:92:E0:6E:03 ManufacturerData Value:
  10 02 53 a0 01 15 91 27 a8 c5 7c 88 f4 7e 5c 98  ..S....'..|..~\.
  37 6f 37 27                                      7o7'
[CHG] Device 4A:D7:A3:21:B1:CD RSSI: -55
[CHG] Device 8C:FE:57:12:07:A1 RSSI: -42
[CHG] Device E4:EE:92:E0:6E:03 RSSI: -72
[CHG] Device E4:EE:92:E0:6E:03 ManufacturerData Key: 0x02e1
[CHG] Device E4:EE:92:E0:6E:03 ManufacturerData Value:
  02 e3 53 5c 54 99 72 dc 3b 03 e6 ae 9c 0b 11 8b  ..S\T.r.;.......
  7b 27 9a b7 88 21 31 3d 9b 60 6e                 {'...!1=.`n
[CHG] Device 31:64:15:1A:47:C0 RSSI: -46
[CHG] Device 31:64:15:1A:47:C0 RSSI: -47
[CHG] Device 31:64:15:1A:47:C0 RSSI: -45
[CHG] Device 31:64:15:1A:47:C0 RSSI: -48
[CHG] Device E4:CC:56:AD:19:D2 RSSI: -38
[CHG] Device 8C:FE:57:12:07:A1 RSSI: -45
[CHG] Device 4D:57:16:E4:FF:F8 RSSI: -39
[CHG] Device 31:64:15:1A:47:C0 RSSI: -43
[CHG] Device 8C:FE:57:12:07:A1 RSSI: -44
[CHG] Device 31:64:15:1A:47:C0 RSSI: -46
[CHG] Device 31:64:15:1A:47:C0 RSSI: -46
[CHG] Device 31:64:15:1A:47:C0 RSSI: -46
[CHG] Device 8C:FE:57:12:07:A1 RSSI: -42
[CHG] Device C8:A9:D1:FF:A1:1A RSSI: -40
[CHG] Device 31:64:15:1A:47:C0 RSSI: -47
[CHG] Device 31:64:15:1A:47:C0 RSSI: -48
[CHG] Device 31:64:15:1A:47:C0 RSSI: -43
[CHG] Device 4D:57:16:E4:FF:F8 RSSI: -38
[CHG] Device 31:64:15:1A:47:C0 RSSI: -48
[CHG] Device 8C:FE:57:12:07:A1 RSSI: -42
[CHG] Device 31:64:15:1A:47:C0 RSSI: -43
[CHG] Device C7:BB:22:B4:C2:1C RSSI: -54
[CHG] Device 4A:D7:A3:21:B1:CD RSSI: -53
[CHG] Device E4:EE:92:E0:6E:03 RSSI: -72
[CHG] Device E4:EE:92:E0:6E:03 ManufacturerData Key: 0x02e1
[CHG] Device E4:EE:92:E0:6E:03 ManufacturerData Value:
  02 e3 53 5c 54 99 7e dc 3b 03 ae c0 8c 50 b3 8e  ..S\T.~.;....P..
  cd 5f 8e c5 39 7d dc d3 a6 a1                    ._..9}....
[CHG] Device 31:64:15:1A:47:C0 RSSI: -45
[CHG] Device CE:CC:A8:96:38:66 RSSI: -55
[CHG] Device 31:64:15:1A:47:C0 RSSI: -45
[CHG] Device 31:64:15:1A:47:C0 RSSI: -45
[CHG] Device 31:64:15:1A:47:C0 RSSI: -46
[CHG] Device 4A:D7:A3:21:B1:CD RSSI: -56
[CHG] Device 31:64:15:1A:47:C0 RSSI: -45
[CHG] Device 4A:04:40:AF:E6:33 RSSI: -46
[CHG] Device 31:64:15:1A:47:C0 RSSI: -43
[CHG] Device 4D:57:16:E4:FF:F8 RSSI: -38
[CHG] Device 31:64:15:1A:47:C0 RSSI: -43
[CHG] Device 31:64:15:1A:47:C0 RSSI: -44
[CHG] Device 8C:FE:57:12:07:A1 RSSI: -44
[CHG] Device 31:64:15:1A:47:C0 RSSI: -45
[CHG] Device A5:C2:37:36:32:A7 RSSI: -52
[CHG] Device A5:C2:37:36:32:A7 Connected: yes
Connection successful

For the secons Connection the result is:

pi@raspberrypi:~ $ bluetoothctl info  A5:C2:37:36:32:A7
Device A5:C2:37:36:32:A7 (public)
        Name: DP04S007L4S100A
        Alias: DP04S007L4S100A
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00010203-0405-0607-0809-0a0b0c0d1912)
        Modalias: usb:v248Ap8266d0001
        ManufacturerData Key: 0x32a7
        ManufacturerData Value:
  36 37 c2 a5                                      67..
        RSSI: -56

naugehyde, 2025-08-20:
FYI If you get a BT dongle like the Toplink UB 500 Plus which I recommend and use on my own boat, you'll be able to connect to both your victron devices and the JBDBMS.

naugehyde, 2025-08-22:
Guten Abend Chris,

The dongle should make a difference. I'll reopen this issue until you're up and running.

FYI it's almost certainly a JBDBMS BMS atop an EcoWorthy battery so choose JBMBMS from the sensor class menu when setting up. The EcoWorthy branded BT-BMS is an add on that looks like:

[image]

The next version of the plugin renames EcoWorthy to EcoWorthyBW02 to help avoid the confusion.

-Andy

L0wbird81, 2025-08-27:
Hey Andy, maybe another hint, 

https://github.com/mike805/eco-worthy-battery-logger

L0wbird81, 2025-08-27:
I tried to select the JBDBMS, that’s what you see in the log. But there are no SignalK paths an there is an exclamation mark at the RSSI Triangle.

naugehyde, 2025-08-27:
Hi Chris. 

It appears to be working up to the point that it sends the command (0x3) to the device to get data. 

Keep the log open. If after 30s you see:

Look for: `Response timed out (+30s) from JBDBMS device <name of your device>`

Any other errors will be useful too of course.

It appears to have worked at least once or the plugin would not have been able to configure the device. (On configuration it queries the device for the number of cells.)

Let me know as soon as you know. Standing by.

naugehyde, 2025-08-27:
The RSSI indicator will show an ! for now (it should show that it's connected -- which it is)