naugehyde/bt-sensors-plugin-sk #80: Victron Orion Xs  missing Signal K path.

SignalKuser, 2025-08-31:
I have 3 victrons ,Mppt and Charger Paths are  correct and complete  in signal K. but the orion xs path in Signal K isnt there. only the orion RSSI Path (db).

[image]
[image]
[image]
[image]

how can i solve the problem and show me the missing orion Paths ?
thx , chris

naugehyde, 2025-08-31:
Happy to help.

Please: 

1) Turn on debug for the plugin 
2) restart the plugin and report any errors in the server log
3) attach your config file here

4) as an experiment, temporarily turn off the jbd plugin -- there may be a conflict between the two plugins (note: the bt-sensors plugin does support the JBD device)

naugehyde, 2025-08-31:
On reflection, it's very likely that the JBD plugin is conflicting with the BT-Sensors plugin. 

There's a quirk in some Linux systems where requesting a connection to a device (like the JBD BMS),  the underlying Bluetooth system turns off the Bluetooth scanner without notification to other apps. 

Applications reading Victron devices require that the scanner is on in order to receive data. They are not connected devices but passive emitters of data like many other BT sensor devices. If the scanner is off, no data will be received by Signalk except by connected devices. 

Your screenshots indicate that other Victron devices (the AC Charger for example) are not receiving data either. The Orion just had the bad luck of not being started until the JBD device had started and thereby turned off the scanner. The Out Of Range indication on the OrionXS very likely resulted from the scanner being off when the plugin attempted to locate the device. 

NOTE: My plugin accounts for the scanner-off quirk by turning on the scanner after making a connection.

There are a number of things you can try in addition to the diagnostics recommended earlier:

1) Use the BT-Sensors plugin and not the JBD-overkill-BMS to communicate with the JBD battery. 
2) Disable and re-enable the BT-Sensors plugin _after_ the JBD-overkill-BMS has connected to the JBD BMS.

Best,

Andy

PS If you want to use the BT-sensors plugin, you will want to upgrade to the latest Beta which is available only on NPM. To install it, open a command line on your server and go to your Signalk home (usually ~/.signalk) and execute the following: 

`npm i bt-sensors-plugin-sk@1.2.6-beta-3`

PPS The JBD-overkill-BMS plugin does a better job of displaying the protection status for the BMS. I'll mirror those capabilities for the JBDBMS class in my plugin in the next release.

SignalKuser, 2025-09-01:
thx so far, will be on my boat in  two weeks and  try to fix the problem. in one screenshot  you see the 3 victrons, 2 are deactivated. i will try your plugin for my batterie. months before the overkill plugin was the only plugin can see my batterie. first i tried d-bus serial, that failed, then signal k and the plugins, only overkill works. but i will give a try, batterie is a basengreen 12V 230A , BMS have only BLuetooth. by the side: the two other victrons (charger and Mppt) works fine with the overkill together. no Problems.

thank you, will send error log ect.  in 2 weeks.
greets/chris.

SignalKuser, 2025-09-16:
cause bad weather i go wednesday to the boat. then i will check the solution 1: deactivete JBD overkill plug, 2: deinstall JBD overkill plug, 3: try the Beta 
the timestamp thing happen, when you deactivate the victronstuff (charger and mppt) in your plugin, what you can see in Pic2, but not restart signal k and make a screenshot. victron updates are actuell, since i check sometimes with smartphone app.
but i have a another question too: both, the mppt and charger names begin with Victron. ex: VictronACCCharger.Error.Code. why OrionXS begin with "electrical" and not with Victron too? can there be a conflict, JBD overkill use electical inital name too....

greets/Chris

SignalKuser, 2025-09-16:
Ohh, i see you are on version 1.3.0
shoud i use  1.2.6 Beta 3 oder dierctly the 1.3.o Version?

greets/chris

naugehyde, 2025-09-16:
Yes, please use the latest AppStore version.

The "id" parameter in the plugin config is used to distinguish one charger's output from another. You can set the value to anything-- aft, port, house, starter etc.

SignalKuser, 2025-09-18:
[image]

SignalKuser, 2025-09-18:
Ok. i have seen the field for the ID in Orion. but i dont know what value come in there. where i can find the ID parameter for the Field?  
I searched the app but found nothing with ID. 

greets. Chris

SignalKuser, 2025-09-18:
yes, i see both fields. but dont know what values come in. model ID?

naugehyde, 2025-09-18:
Set batteryID to whatever value you want to see: starter, house etc.

SignalKuser, 2025-09-18:
Ahhh. Understand.

SignalKuser, 2025-09-21:
hi, 
ii enabled the debug for orion:
"VictronOrionXS (Model ID: 41968) (F6:83:84:89:71:1A RSSI: -64 db / 62%) ▂ ▄ ▆
Sep 21 17:45:47 Delta is missing value in {"path":"electrical.chargers.OrionXS.offReason"} from source "Victron VictronOrionXS (Model ID: 41968)"
the field is empty.

with another plugin the value for off reason is "remote_input|engine_shutdown" when the engine is off, when engine is on " remote_input|engine is running"
is it possible to implement this feature? 
the remote_input thing is a wire from alternator to orion, it checks that engine is running what is very importent. 

thx.greets/chris

naugehyde, 2025-09-21:
That's a bug. Not sure why it's happening though. 

I'll send some diagnostic code later today when has a chance to go to my boat. You'll need to install it from a command line. I hope your comfortable with that.

naugehyde, 2025-09-21:
Found the issue. Look for fix in next release.