naugehyde/bt-sensors-plugin-sk #82: New device request: ShellyPro3EM

JulianBuss, 2025-09-14:
The ShellyPro3EM monitors 230V power lines and reports current and watthours. Any chance of supporting this handy little gadget in your plugin?

JulianBuss, 2025-09-14:
seems so, I can see it in the plugin!

sebastianhaas, 2025-10-01:
Alright, I did a quick research now. It seems like the Shelly Pro 3EM does not advertise BTHome data, but only acts as a gateway for other sensors that publish BTHome-formatted-data and the forwards it via its own interfaces. But that is just what I understood from a quick look at the available documentation.

You could try and https://play.google.com/store/apps/details?id=cloud.shelly.bledebug and use this app or maybe even https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp to learn about the actual advertisment data.

However, looking at the values you provided and comparing them with the BTHome format https://bthome.io/format/, I don't think it supports these kind of use cases. You will instead be much better off integrating the Shelly Pro 3EM with SignalK using WiFi.