naugehyde/bt-sensors-plugin-sk #85: rpi 3 continuous scanning

isarychkin, 2025-09-17:
bluetoothctl in terminal on rpi's 3 onboard hci0 shows continuous scanning and reporting RSSI, which can only be stopped by disabling the plugin in signalk.  The plugin's "Scan for new devices interval" is set to 0.

rpi 3 bullseye
SK v. 2.16
nodejs v. 22 lts

Cheers.

closed 2025-09-17 by naugehyde

isarychkin, 2025-09-17:
> That's working as designed.
> 
> The system scanner needs to be on to pick up broadcasts from non-connected devices like Victron instant readout devices, RuuviTags etc.
> 
> The scan for new devices refers to whether the plugin displays previously unconfigured devices.

Thanks very much, all clear now.  The scanning process seems quite intense though.  Is it possible to limit the background scans only to devices/MACs of interest?

naugehyde, 2025-09-17:
The plugin is doing the filtration itself. 

There isn't filtration at the system level that works for multiple mac addresses -- just UUIDs and one mac pattern. 

What if any benefit is there for you if there was a more comprehensive filter?