naugehyde/bt-sensors-plugin-sk #86: v1.3.2 Victron Smart BatteryProtect Error

Peter-Petrik, 2025-09-27:
After upgrading from v1.3.0 to v1.3.2 I get the following error:

`Victron Smart BatteryProtect 12/24V-100A D4:1A:5B:xx:xx:xx) Unable to activate sensor. Reason: Unable to read data from Victron Smart BatteryProtect 12/24V-100A (D4:1A:5B:xx:xx:xx RSSI: -76 db / 42%) ▂ ▄ : Error: Route.notify() requires a callback function but got a [object Number]`

naugehyde, 2025-09-27:
Yeah. Regression error. My bad.

to patch run the following from a command line:

`wget https://raw.githubusercontent.com/naugehyde/bt-sensors-plugin-sk/refs/heads/main/sensor_classes/VictronSmartBatteryProtect.js -O ~/.signalk/node_modules/bt-sensors-plugin-sk/sensor_classes/VictronSmartBatteryProtect.js`

Then restart.

PS Two important projects this autumn/winter for me are to normalize and fix the sensor alarms and to create some kind of testing platform for all the various sensors -- most of which I do not have on hand.

Peter-Petrik, 2025-09-28:
Unfortunately I'm getting the same error after obtaining the patched file and restarting.

naugehyde, 2025-09-28:
Unable to recreate but just posted a small change to the VictronSmartBatteryProtect file that might make a difference.

Re-download and try again please.

closed 2025-09-28 by naugehyde